cocotb/cocotb #182: Deadlock between coroutines

nfrench-intellisis, 2015-01-05:
Hi,

I have been using cocotb for a few weeks.  I ran into what seems like a thread deadlock when passing data between coroutines using the below code.  With the log.info in wait_for_packet() everything works fine.  When I remove it it seems that the packet_monitor never gets a chance to run.

I've stripped this down into example code as best I can.  I haven't included the design source code.  If that's required I will get an RTL guy to concoct a more complete example.

The monitor code:

    """Packet interface monitor."""

    import cocotb

    packets = []

    @cocotb.coroutine
    def wait_for_packet(dut):
        """Waits for packets collected by packet_monitor."""

        while not packets:
            dut.log.info("tick") # <------ without this it doesn't work
            yield RisingEdge(dut.u_top.clk)

    @cocotb.coroutine
    def packet_monitor(dut):
        """Packet monitor."""

        dut.u_top.log.info("starting monitor")
        while 1:
            yield RisingEdge(dut.u_top.start_of_packet)
            data = []

            while not dut.u_top.end_of_packet.value:
                yield RisingEdge(dut.u_top.clk)
                data.append(dut.u_top.data.value)

            packets.insert(0, data)
            dut.log.info("received %s" % data)

Usage:

    import cocotb
    import monitor

    @cocotb.test()
    def test(dut):
        # startup clocks, etc

        # spawn monitor
        cocotb.fork(monitor.packet_monitor(dut))
        yield monitor.wait_for_packet(dut)
        dut.log.info("got a packet")

stuarthodgson, 2015-01-05:
HI,

How do you know that the packet_monitor is not running? Rather than dut.u_top.start_of_packet going high? Can you insert the following into the packet_monitor 

```
@cocotb.coroutine
def packet_monitor(dut):
    """Packet monitor."""

    dut.u_top.log.info("starting monitor")
    while 1:
+++     dut.log.info("In Monitor Loop")
+++     yield RisingEdge(dut.u_top.clk)
        yield RisingEdge(dut.u_top.start_of_packet)
        data = []
```

I used your code against a small piece of RTL and it worked as expected. However with no dut.log.info it appears to do nothing until I set `dut.u_top.start_of_packet <= 1`

nfrench-intellisis, 2015-01-05:
Adding the dut.log.info call in the packet_monitor as you describe eliminates the issue, just as adding it to the other coroutine eliminates the problem.

Is the way I am passing data (using a python list as a queue) between these two contexts valid?  Is there a better way?  Lists are thread safe so it seemed like the thing to do.

Should I put together a mockup that includes RTL and testbench so you can just run it to reproduce the issue?  And thanks for the support.

chiggs, 2015-01-06:
Note you may be interested in subclassing the [Monitor class](https://github.com/potentialventures/cocotb/blob/master/cocotb/monitors/__init__.py) (or the `BusMonitor` class in the same file) which provide some common capabilities.  You'd basically override the `_monitor_recv` method.  Your code would look something like this:

```
import cocotb
from cocotb.monitors import Monitor
from cocotb.triggers import RisingEdge, ReadOnly

class MyMonitor(Monitor):

    def __init__(self, dut):
        self.dut = dut
        Monitor.__init__(self)

    @cocotb.coroutine
    def _monitor_recv(self):
        while True:
            yield RisingEdge(self.dut.clk)
            yield ReadOnly()
            if self.dut.u_top.start_of_packet:
                data = []
                data.append(self.dut.u_top.data.value)
                while not self.dut.u_top.end_of_packet.value:
                    yield RisingEdge(self.dut.u_top.clk)
                    yield ReadOnly()                    
                    data.append(self.dut.u_top.data.value)
                self._recv(data)
```

Then in your test you could do the following:

```
mon = MyMonitor(dut)
packet = yield mon.wait_for_recv(timeout=100000)
```

You can also get verbose logging by doing `mon.log.setLevel(logging.DEBUG)`, register a callback for when packets arrive etc.  There are further examples of usage with the [Avalon Monitors](https://github.com/potentialventures/cocotb/blob/master/cocotb/monitors/avalon.py).

Regarding communication between coroutines - Cocotb provides two mechanisms:

1. The `Event` trigger:

```
def some_coroutine(event):
    yield Timer(100000)
    e.set("Bing")

e = Event()
cocotb.fork(some_coroutine(e))
yield e.wait()
print "Got: %s" % e.data
```

Note that multiple coroutines can be waiting on the event, although the order in which they will be scheduled is not guaranteed.

2. The `ReturnValue` exception

Unfortunately the use of coroutines for generators makes it difficult to return a value from a `yield`, so we've used an exception to achieve this:

```
def some_coroutine():
    yield Timer(100000)
    raise ReturnValue("Bing")

got = yield some_coroutine()
print "Got: %s" % got
```

Disclaimer - the code above is just typed from memory and hasn't been tested. There will be typos / errors but the general gist should be correct...

However the mechanism you have used should work, and I don't understand why adding a log statement would make any difference so we should investigate further.

If you are able to provide a simple testcase that would be exceedingly helpful.

nfrench-intellisis, 2015-01-06:
I'll look into providing a real test case but it may be a while before we can do that.

I experimented with the debug logging in the scheduler (using COCOTB_SCHEDULER_DEBUG environment variable).  It also masks the problem.  I mucked with the scheduler to have it log to memory instead of stdio and it still masks the problem.

Also it turns out the problem is intermittent.  Because of this I'm having trouble instrumenting the schedule and catching the problem.  As I understand your scheduler more I'm more confused why logging would mask the issue.

I will play with the event or ReturnValue mechanisms, they may be a cleaner way to do things anyway.

Will get back to you guys when I know more or can provide a real test case.

stuarthodgson, 2015-01-06:
What logging does do though is call down into the sim to get the time to display in the log message. I know this all sounds a bit trial an error but this is not something that we have seen.

Could you try replacing the dut.log.info with a basic print statement?

nfrench-intellisis, 2015-01-07:
Last night I tried against the latest master and no difference.

My test looks like this:

    cocotb.fork(test_clock.clock_gen(dut.clk))
    cocotb.fork(pkt_monitor(dut))
    yield send_pkt(dut, pkt)
    rsp = yield wait_for_packet(dut)

ktbarrett, 2020-04-23:
This issue is quite old, if you are still having an issue, please reopen.

closed 2020-04-23 by ktbarrett