cocotb/cocotb #186: Python 3 support

jeremyherbert, 2015-01-11:
Hi,

As I love the idea of cocotb and you were nice enough to open source it, I'd like to give something back. So I've started porting the code to python3 (specifically targeting python 3.4 which is the latest version). It's going well so far, I've built the module and am currently debugging a simulator crash with iverilog. I will update this issue with any progress I make.

The code so far can be found in my fork: https://github.com/jeremyherbert/cocotb
It's not ready for a pull request, because it breaks python 2.7 support. However, this is only because I'm trying to get python3 working first. I will put the changes into a tidy pull request when I am satisfied with it working.

Any comments or suggestions would be greatly appreciated. I will be working on this on and off over the next few days.

stuarthodgson, 2015-01-12:
HI Jeremy,

First of all thanks for this work, as Chris mentioned it is something that
we have had on the to do list for a while do really grateful that you are
taking it on.

As for any general comments, It would be nice if any python specific things
in the simulator module could be pushed to an implementation file and
restrict #ifdefs to headers only.

So have something like

simulatormodule.h
#ifdef PY_MAJOUR_VERSION 3
..
#ifelse
..
#endif

And then have
simulatormodule_py3.c
simulatormodule_py2.c

which are included in the build as needed.

So maybe doing a small abstraction layer for the PyString or PyUnicode in a
common headers as well.

Stu

jeremyherbert, 2015-01-12:
Just an update on this, I have now simulated a very simple adder using iverilog and python3 and it works fine. Most of the tests are still broken because they use python2 syntax, but I will work on that next.

I plan on keeping as many macros out of the c files as possible, but unfortunately some are unavoidable (like changing the name of the shared module initialisation function).

jeremyherbert, 2015-01-12:
@stuarthodgson 

Can you please post a log of a working "make" using iverilog? I'm having some issues where an exception in a test for 2.7 is taking down the whole simulator.

chiggs, 2015-01-12:
@jeremyherbert There's a full log of a successful build on Travis: https://travis-ci.org/potentialventures/cocotb

jeremyherbert, 2015-01-12:
Thanks chris. On that page, line 308 has quite a peculiar error, is that supposed to be there? I actually thought that was my fault, but it can't be in that build.

chiggs, 2015-01-12:
@jeremyherbert Some of the tests are supposed to fail and so will report errors.  In this particular case we do expect the test to fail, but it's actually failing for the wrong reason so I'll take a look.

However in general don't be alarmed by the output of `make tests` - we should really change the default rule to run an example and not all the tests as it can be off-putting to see errors in the output.

stuarthodgson, 2015-01-12:
@jeremyherbert
da0cb829 Should resolve this

jeremyherbert, 2015-01-12:
Ok perfect. Thanks.

I'm almost done here, I have one last problem with the endian_swapper example in python3 which I can't seem to nail down. I'll give it a shot tomorrow. If you get a chance, it would help if you could try to build my master branch on one of your systems too with python 2.7.* and 3.4.*? See if the tests pass.

Unfortunately I can't test with any simulators other than iverilog because I'm a poor student :) And even if I wasn't, some of them still cost more than a car!

Some notes for future compatibility: 
- don't use `generator.next()`, use `next(generator)`
- if you are going to print in a file, use `from __future__ import print_function` at the top and `print("hello")`
- `xrange()` doesn't exist in python3, because it has been renamed to `range`. So probably not a good idea to use it anymore
- python `long` doesn't exist anymore, all integers are longs
- there is a bug which has not been fixed in 2.7 which means that tracebacks are not presented as unicode. This is the only non-backwards compatible python problem that I found. There is unfortunately version specific code for this part. See http://bugs.python.org/issue6543 for more

Given that guido has put his foot down and said no python 2.8, it would be great if you guys could see if a separate travis-ci run could be set up with python3.4. What do you think?

jeremyherbert, 2015-01-13:
Ok, so travis-ci breaks on the 3.4 build because of this bug: https://github.com/pypa/virtualenv/issues/169 I have put a workaround in the .travis.yml script.

jeremyherbert, 2015-01-13:
Ok, so thanks to the help of @themperek, we are almost there. Travis-CI builds pass for python2.7, and there is one test group failing for python3.4: https://travis-ci.org/jeremyherbert/cocotb/builds/46848585

@chiggs or @stuarthodgson, could you please take a look at the currently failing test? Unfortunately I don't have a whole lot of experience debugging with verilog.

chiggs, 2015-01-13:
@jeremyherbert If you merge current master (with b21470e) onto your branch I think P3K should work.  There were a few other incompatibilities, the main one causing the failure was the change from `__nonzero__` to `__bool__`.

jeremyherbert, 2015-01-13:
We made it! https://travis-ci.org/jeremyherbert/cocotb/builds/46920608

I'm going to sleep on it now and I will set up the pull request tomorrow. Do you mind if I have many commits? or do you want me to try and squash them ?

potentialventures, 2015-01-13:
Great news,

I don't want to be a pain but if it is not too much trouble could you try
and squash them down. So group commits into, travis support, build changes,
syntax changes and such.

Stu

themperek, 2015-01-14:
Hi,

It is maybe not appropriate place so sorry.
I use non standard python installation (anaconda). I also have to set some variables before. Maybe better would be to do this in a makefile?
```bash
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:$(python -c 'from distutils import sysconfig; print(sysconfig.get_config_var("LIBDIR"))')
export PYTHONPATH=$PYTHONPATH:$(python -c "from distutils import sysconfig; print(sysconfig.get_python_lib())"):$(COCOTB)
export PYTHONHOME=$(python -c "from distutils.sysconfig import get_config_var; print(get_config_var('prefix'))")
```
Tomek

jeremyherbert, 2015-01-14:
Hi @themperek,

I agree that should be changed, but I think I'm going to leave it out of the python3 commits. They are large enough already, and it should be easy to fix with another pull.

Doing the squash now.

jeremyherbert, 2015-01-14:
Ok, looks like this one is sorted. Thanks to everyone for your assistance.

closed 2015-01-14 by jeremyherbert