cocotb/cocotb #196: CVC support

chiggs, 2015-01-19:
We should support a CVC from tachyon: http://www.tachyon-da.com/

jeremyherbert, 2015-01-21:
I'd like to give this one a shot.

I'm going to send an email to the CVC guys to see if it can be hosted publicly somewhere so that it can be installed during the travis builds. Will report back.

themperek, 2015-01-22:
I have send this already and received the answer.

Some small changes (https://github.com/themperek/cocotb)  and endian_swapper seems to work but only in interpreted mode.
You have to use CVCx : https://github.com/themperek/cvcx (if someone is interested in taking care of this or have idea for a better name?)

Debug seems to work well.

themperek, 2020-04-16:
It is supported to some extend (simple example works).  Beyond that, it is CVC issues.  
Code available here: https://github.com/cambridgehackers/open-src-cvc

closed 2020-04-16 by themperek