cocotb/cocotb #203: SystemVerilog clocking like construct/method

tjora, 2015-01-22:
Hi,
I find it nice to do sample signals from the DUT right before the clock edge goes high and update signals to the DUT some delay after the clock edge. This makes the TB quite robust, and it could also be used to test synthesised design with back-annotated timing. In SystemVerilog one could use a clocking-block to achieve this. This is one method to simulate setup and hold times.

Is there any any recommended way of doing this in cocotb?

Currently I hack the delayed output signals by a
 yield Timer(10)
before the signal assignments.

I have not found any sensible way to do pre-clockedge sampling though. Any ideas?

Sigve

chiggs, 2015-01-23:
I think it should be possible to have a ClockingBlock like construct that looked like an actual clock.  This could be created with a sample phase offset and a drive phase offset and the main part would be a coroutine consisting of a few timers around the RisingEdge of a clock.

We would have to create a fake RisingEdge trigger to wake up pending Drivers/Monitors. This clock could then be passed to a Monitor or Driver instance as the synchronous clock.

The difficult part is distinguishing between a sample phase and a driving phase.  As you note in #204 we have used `ReadOnly` to do this, however perhaps we need to create a new Trigger class (`SamplePhase` or similar).

In the simple case this could be `ReadOnly` to match the existing behaviour, but in the more complex case it could be a chain of triggers (i.e. a `RisingEdge` followed by a `Timer`).  Unfortunately I don't think there's a neat way to do this without changing the existing drivers although it should remain backwards compatible as ReadOnly would still work as before.

Needs a bit more thought and discussion.

congeal, 2015-04-06:
How's going with this issue?
I used to drive a clock in Cocotb testbench until a logic sampled a value from a signal right after that was assigned on the rising edge of a clock.
I supposed that the new value assigned on a clock edge were read at the next rising edge but it was sampled at the time the value was assigned.
It seems that this is happens when the clock signal and the data signal are updated at the same time.
(I heard that it is why Systemverilog clockingblock struct was born.)
So, I have to drive a clock in Verilog side to get the logic to sample the value on the next edge.
(Sure, I have to make a Verilog testbench wrapping a DUT and including a clock driver instead of using the DUT as Verilog top module.)
Do you have any plan to resolve this problem?

ktbarrett, 2021-08-23:
Easy to accomplish using `RisingEdge(dut.clk)` with a higher-order task function for doing delayed writes. cocotb doesn't need to provide this functionality.

```python
def delayed_assignment(
    handle: SimHandleBase,
    time: Tuple[int, str],
    value: Any
) -> RunningTask:
    async def writer():
        await Timer(*time)
        handle.value = value
    return cocotb.fork(writer())
```
   
Yes, it is higher overhead than handling delayed writes in the write flush task in the scheduler, but that is a separate discussion on optimization.

> Currently I hack the delayed output signals by a
yield Timer(10)
before the signal assignments.

This is not a hack, this is correct.

> I have not found any sensible way to do pre-clockedge sampling though.

Unnecessary, `RisingEdge` leaves you before sympathetic signals change. `await RisingEdge(clk)` and then doing writes is exactly equivalent to a clocked process in HDL.

closed 2021-08-23 by ktbarrett