cocotb/cocotb #204: Use of ReadOnly() phase

tjora, 2015-01-22:
Hi,
I do not understand why signals in the example code often are sampled in the ReadOnly phase. Take for example _send_write_address  from AXI4LiteMaster ( from https://github.com/potentialventures/cocotb/blob/master/cocotb/drivers/amba.py)

```python
    @cocotb.coroutine
    def _send_write_address(self, address, delay=0):
        """
        Send the write address, with optional delay
        """
        yield self.write_address_busy.acquire()
        for cycle in range(delay):
            yield RisingEdge(self.clock)

        self.bus.AWADDR         <= address
        self.bus.AWVALID        <= 1

        while True:
            yield ReadOnly()
            if self.bus.AWREADY.value:
                break
            yield RisingEdge(self.clock)
        yield RisingEdge(self.clock)
        self.bus.AWVALID        <= 0
        self.write_address_busy.release()
```

Here AWADDR and AWVALID are updated on a RisingEdge of the clock. The signal AWREADY is however sampled in the ReadOnly() phase. The AWREADY could have been updated by some clocked always-block in the design between the clock edge and the start of the ReadOnly phase. Thus AWREADY is effectively sampled one clock cycle early (compared to how it would be sampled by an always_ff block in SystemVerilog). However, if the model is used to simulate a model with backannoteted timing, the timing will be different. Here the while True: loop will run one more time, causing no real difference, but in other cases the testbench may break.

What is the rationale for sampling the AWREADY.value in the ReadOnly phase on not just right after the yield RisingEdge(self.clock) statement?

I should probably add that this may be a (system)verilog issue, as I have not investigated how ReadOnly maps to VHDL simulation semantics.

Sigve

mzabeltud, 2016-04-07:
I agree with the sampling on the rising edge of the clock because waiting for the ReadOnly() phase will not work if the delay of outputs is changed to non-zero values. Apart from simulation back-annotated models, non-zero output delays are also found in third party IP cores.

I'm not familiar with the timing in SystemVerilog. But, I want to point out that the [Avalon BusMonitor](https://github.com/potentialventures/cocotb/blob/master/cocotb/monitors/avalon.py) waits for the `RisingEdge()` before the `ReadOnly()` phase:

    def _monitor_recv(self):
        """Watch the pins and reconstruct transactions"""

        # Avoid spurious object creation by recycling
        clkedge = RisingEdge(self.clock)
        rdonly = ReadOnly()

        # NB could yield on valid here more efficiently?
        while True:
            yield clkedge
            yield rdonly
            if self.bus.valid.value:
                vec = self.bus.data.value
                self._recv(vec.buff)

Maybe, there is just some inconsistency?

I have used waiting for `RisingEdge()` **only** in all (3) of my complex testbenches so far as well as in this proposed simple D-FF example #427. This example also demonstrates that one can check the initial values of resisters on the first rising edge of the clock which would not be possible when the (output) monitor also waits for `ReadOnly()`.

ktbarrett, 2021-01-30:
This seems like a question that was answered, and no identifiable bug was found. Reopen in [cocotb/cocotb-bus](https://github.com/cocotb/cocotb-bus) if unanswered or a bug is found in the AXI Drivers.

closed 2021-01-30 by ktbarrett