cocotb/cocotb #205: Inconsistent behavior for set_signal_value with int and string arguments

tjora, 2015-01-22:
In VpiCbHdl.cpp
int VpiSignalObjHdl::set_signal_value(int value)
schedules the update with vpiInertialDelay, delay 0

while
int VpiSignalObjHdl::set_signal_value(std::string &value)
schedules the update vpiNoDelay

I would assume that both values should be scheduled with vpiInertialDelay, delay 0.

I am not sure if there is a difference in practice between these two methods. In the SystemVerilog specification there are used different wordings for these two cases, but I do not know how they are implemented in different simulators.

tjora, 2015-01-22:
For the VhpiCbHdl.cpp I think the
vhpi_put_value(.., vhpiForcePropagate)

should be replaced with a call to 
vhpi_schedule_transaction with delaymode set to vhpiInertial
for consistency with the verilog call.

tjora, 2015-01-22:
Is this the reason for the use of setcachedvalue and the save_write infrastructure as well? I think that if the writes are done with vpiInertialDelay, the writes will not be active before the simulator has run to the next phase even with delay of 0, but here I may be very well be mistaken.

Sorry for the noise, I am trying to understand this.

eric-wieser, 2019-03-20:
This was fixed in #792, since an actual bug was found that this caused.

closed 2019-03-20 by themperek