cocotb/cocotb #209: Icarus 'assign' statements don't work

jeremyherbert, 2015-02-06:
Probably related to #96 .

http://www.edaplayground.com/x/4nt
http://www.edaplayground.com/x/96k

Given a very simple and (apparently) identical testbench written in verilog and in python, the output is different if there is an 'assign' style asynchronous assignment. Run the code and look at the EPWave window to see the output.

Can someone please run these under a non-icarus simulator to see if the problem is in icarus or in cocotb?

chiggs, 2015-02-06:
I need to check in more detail, there is a bug in Icarus which is fixed in newer versions which meant that values assigned outside of a process weren't updated.

chiggs, 2015-02-06:
I can confirm this works on Modelsim and Riviera so it is the Icarus bug.

You could try either compiling the latest version of Icarus, or switch to the free version of Modelsim.

potentialventures, 2015-02-08:
Not a bug in Cocotb, closing.

closed 2015-02-08 by potentialventures