cocotb/cocotb #217: Unable to generate libvpi.so file via questa

krishnasmtvs, 2015-02-26:
We run the Cocotb examples in redhat 5 with python 3.4.2, shared object file "libvpi.so" is not generating if we run with, make SIM=questa

But via IUS and icarus, libvpi.so is generating, once lipvpi.so file generate with IUS, we tried to run with questa as-- make SIM=questa

we got an error as,

QuestaSim-64 qverilog 10.3d_1 Compiler 2014.10 Oct 25 2014
Start time: 11:34:59 on Feb 26,2015
qverilog -64 -R -pli "/cocotb/build/libs/x86_64/libvpi.so" - "+acc" /home/krishna/cocotb/examples/adder/tests/../hdl/adder.v
-- Compiling module adder

Top level modules:
        adder
Reading pref.tcl

# 10.3d_1

# vsim -lib work -pli "/cocotb/build/libs/x86_64/libvpi.so" adder -c -do "run -all; quit -f" -appendlog -l qverilog.log -vopt
# ** Note: (vsim-8009) Loading existing optimized design _opt
# Loading /cocotb/build/libs/x86_64/libvpi.so
# ** Error: (vsim-3197) Load of "/cocotb/build/libs/x86_64/libvpi.so" failed: /cocotb/build/libs/x86_64/libvpi.so: cannot open shared object file: No such file or directory.
# ** Error: (vsim-PLI-3002) Failed to load PLI object file "/cocotb/build/libs/x86_64/libvpi.so".
#         Region: /
# //  Questa Sim-64
# //  Version 10.3d_1 linux_x86_64 Oct 25 2014
# //
# //  Copyright 1991-2014 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  THIS WORK CONTAINS TRADE SECRET AND PROPRIETARY INFORMATION
# //  WHICH IS THE PROPERTY OF MENTOR GRAPHICS CORPORATION OR ITS
# //  LICENSORS AND IS SUBJECT TO LICENSE TERMS.
# //
# Loading work.adder(fast)
# Error loading design
Error loading design
# End time: 11:35:06 on Feb 26,2015, Elapsed time: 0:00:07
# Errors: 2, Warnings: 0
make[1]: *** [results.xml] Error 12
make[1]: Leaving directory `/cocotb/examples/adder/tests'
make: *** [sim] Error 2

Can you please give solution for this issue

stuarthodgson, 2015-02-26:
We do indeed not seem to build libvpi.so when using questa. I'll push a fix shortly.

stuarthodgson, 2015-02-26:
@krishnasmtvs 

If you update to 9d3f44e7.. then VPI should now be present under Questa. Please get in touch with any more issues.

Stu

krishnasmtvs, 2015-02-26:
I made change in makefiles/simulators/Makefile.questa as

results.xml: $(SIM_BUILD) $(VERILOG_SOURCES) $(CUSTOM_COMPILE_DEPS) $(CUSTOM_SIM_DEPS) $(COCOTB_LIBS) $(COCOTB_LIB_VPI)

to

results.xml: $(SIM_BUILD) $(VERILOG_SOURCES) $(CUSTOM_COMPILE_DEPS) $(CUSTOM_SIM_DEPS) $(COCOTB_LIBS) $(COCOTB_VPI_LIB)

Now libvpi.so is generating but

if I run as make SIM=questa
Now I got error as,

Error: (vsim-3197) Load of "/home/krishna/cocotb/build/libs/x86_64/libvpi.so" failed: /home/krishna/cocotb/build/libs/x86_64/libcocotb.so: undefined symbol: _Py_NoneStruct.
# ** Error: (vsim-PLI-3002) Failed to load PLI object file "/home/krishna/cocotb/build/libs/x86_64/libvpi.so".

stuarthodgson, 2015-02-26:
This may be an issue with the python installation you have. Since redhat 5 does not come with python3.5 how did you install it?

Could you run

$> ldd /home/krishna/cocotb/build/libs/x86_64/libcocotb.so

And post the output back here.

krishnasmtvs, 2015-02-26:
Output of ldd is

linux-vdso.so.1 =>  (0x00007fffde9bb000)
        libgpilog.so => /home/krishna/cocotb/build/libs/x86_64/libgpilog.so (0x00002accda60a000)
        libcocotbutils.so => /home/krishna/cocotb/build/libs/x86_64/libcocotbutils.so (0x00002accda80d000)
        libc.so.6 => /lib64/libc.so.6 (0x00002accdaa33000)
        /lib64/ld-linux-x86-64.so.2 (0x0000003b4ac00000)

Default python version in redhat 5 is 2.4.3

But in cocotb documents i checked, its supports python 2.6 + above thats why am installed python3.4.2

krishnasmtvs, 2015-02-26:
If I run via questa only, I got this issue. In cadence I didn't get it any error.

stuarthodgson, 2015-02-26:
The dependencies should be

        linux-vdso.so.1 =>  (0x00007fffeabfe000)
        libpython2.7.so.1.0 => /usr/lib/x86_64-linux-gnu/libpython2.7.so.1.0 (0x00007f2763b7e000)
        libgpilog.so => not found
        libcocotbutils.so => not found
        libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f27637b7000)
        libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f2763599000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007f276337f000)
        libdl.so.2 => /lib/x86_64-linux-gnu/libdl.so.2 (0x00007f276317b000)
        libutil.so.1 => /lib/x86_64-linux-gnu/libutil.so.1 (0x00007f2762f78000)
        libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f2762c71000)
        /lib64/ld-linux-x86-64.so.2 (0x00007f2764313000)

So it does not seem that your system is pulling python in correctly.

You said that the build works with Icarus? Does the test run?

Can you run
$> python -V

Stu

chiggs, 2015-02-26:
Differences between Makefile.ius and Makefile.questa:

PYTHONPATH is the same

Questa:
`LD_LIBRARY_PATH=$(LIB_DIR):$(PYTHON_LIBDIR):$(PYTHON_DYNLIBDIR):$(LD_LIBRARY_PATH)`

vs

IUS:
`LD_LIBRARY_PATH=$(LIB_DIR):$(LD_LIBRARY_PATH)`
`LD_RUN_PATH=$(PYTHON_LIBDIR):$(LD_RUN_PATH)`

Everything else should get picked up from a common Makefile.  Can't see why there would be any difference although these variables need harmonising across the simulator Makefiles.

Is Questa definitely running in 64-bit mode?

stuarthodgson, 2015-02-26:
I don't think that python is being picked up properly at build time, maybe 2.4 is in use rather than 3 even though 3 is installed. As such libsim.so does not link against python and neither does libcocotb.so. At lib load time there is then an unresolved symbol. We did do a fix that would take account of a sim using RTLD_NOW rather than RTLD_LAZY but if there is no dynamic lib on 2.4 then this will not help.

The different dlopen options might be the difference between questa and IUS.

@krishnasmtvs 

Could you post the output of

$> make clean
$> make SIM=questa

Thanks

Stu

krishnasmtvs, 2015-02-27:
krishna@krishna functionality]$ cd tests/
[krishna@krishna tests]$ ls
Makefile  test_cocotb.py  test_discovery.py  test_external.py  test_regression.py
[krishna@krishna tests]$ make SIM=IUS
make results.xml
make[1]: Entering directory `/home/krishna/cocotb/examples/functionality/tests'
mkdir -p sim_build
make -C /home/krishna/cocotb/lib/vpi EXTRA_LIBS= EXTRA_LIBDIRS=
make[2]: Entering directory `/home/krishna/cocotb/lib/vpi'
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DVPI_CHECKING -Wstrict-prototypes -Waggregate-return -shared -Xlinker -L/usr/local/lib -o /home/krishna/cocotb/build/libs/x86_64/libvpi.so  /home/krishna/cocotb/build/obj/x86_64/VpiImpl.o /home/krishna/cocotb/build/obj/x86_64/VpiCbHdl.o   -lgpilog -lgpi -lstdc++  -L/home/krishna/cocotb/build/libs/x86_64
ln -sf libvpi.so /home/krishna/cocotb/build/libs/x86_64/gpivpi.vpl
ln -sf libvpi.so /home/krishna/cocotb/build/libs/x86_64/cocotb.vpi
make[2]: Leaving directory `/home/krishna/cocotb/lib/vpi'
LD_RUN_PATH=/usr/local/lib: PYTHONPATH=/home/krishna/cocotb/build/libs/x86_64:/home/krishna/cocotb:/home/krishna/cocotb/examples/functionality/tests: LD_LIBRARY_PATH=/home/krishna/cocotb/build/libs/x86_64:/home/krishna/cocotb/build/libs/x86_64/:/usr/local/lib:/usr/local/lib:/usr/local/lib:/usr/local/lib MODULE=test_cocotb,test_discovery,test_external,test_regression TESTCASE= TOPLEVEL=sample_module \
               irun -64bit -nclibdirname sim_build  -sv -loadvpi /home/krishna/cocotb/build/libs/x86_64/libvpi +access+rwc /home/krishna/cocotb/examples/functionality/tests/../hdl/sample_module.v
irun(64): 14.10-s007: (c) Copyright 1995-2014 Cadence Design Systems, Inc.
file: /home/krishna/cocotb/examples/functionality/tests/../hdl/sample_module.v
        module worklib.sample_module:v
                errors: 0, warnings: 0
                Caching library 'worklib' ....... Done
        Elaborating the design hierarchy:
        Top level design units:
                sample_module
        Building instance overlay tables: .................... Done
        Generating native compiled code:
                worklib.sample_module:v <0x709a49df>
                        streams:   5, words:  1010
        Building instance specific data structures.
        Loading native compiled code:     .................... Done
        Design hierarchy summary:
                                 Instances  Unique
                Modules:                 1       1
                Registers:               3       3
                Scalar wires:            3       -
                Vectored wires:          2       -
                Always blocks:           3       3
                Initial blocks:          1       1
                Simulation timescale:  1ps
        Writing initial simulation snapshot: worklib.sample_module:v
Loading snapshot worklib.sample_module:v .................... Done
ncsim> source /krishnadata/INCISIV14.10.007/tools/inca/files/ncsimrc
ncsim> run
ncsim: *W,RNQUIE: Simulation is complete.
ncsim> exit
make[1]: Leaving directory `/home/krishna/cocotb/examples/functionality/tests'
[krishna@krishna tests]$ ls
irun.log  Makefile  sim_build  test_cocotb.py  test_discovery.py  test_external.py  test_regression.py  waveform.vcd

[krishna@krishna tests]$ ls
irun.log  Makefile  sim_build  test_cocotb.py  test_discovery.py  test_external.py  test_regression.py  waveform.vcd
[krishna@krishna tests]$ rm -rf irun.log sim_build/ waveform.vcd
[krishna@krishna tests]$ ls ../../../build/libs/x86_64/
cocotb.vpi  gpivpi.vpl  libcocotb.so  libcocotbutils.so  libgpilog.so  libgpi.so  libsim.so  libvpi.so  simulator.so
[krishna@krishna tests]$ rm -rf  ../../../build/libs/x86_64/cocotb.vpi
[krishna@krishna tests]$ rm -rf ../../../build/libs/x86_64/gpivpi.vpl
[krishna@krishna tests]$ rm -rf ../../../build/libs/x86_64/libvpi.so
[krishna@krishna tests]$ ls ../../../build/libs/x86_64/
libcocotb.so  libcocotbutils.so  libgpilog.so  libgpi.so  libsim.so  simulator.so
[krishna@krishna tests]$ make SIM=questa
make results.xml
make[1]: Entering directory `/home/krishna/cocotb/examples/functionality/tests'
mkdir -p sim_build
make -C /home/krishna/cocotb/lib/vpi EXTRA_LIBS= EXTRA_LIBDIRS=
make[2]: Entering directory `/home/krishna/cocotb/lib/vpi'
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DVPI_CHECKING -Wstrict-prototypes -Waggregate-return -shared -Xlinker -L/usr/local/lib -o /home/krishna/cocotb/build/libs/x86_64/libvpi.so  /home/krishna/cocotb/build/obj/x86_64/VpiImpl.o /home/krishna/cocotb/build/obj/x86_64/VpiCbHdl.o   -lgpilog -lgpi -lstdc++  -L/home/krishna/cocotb/build/libs/x86_64
ln -sf libvpi.so /home/krishna/cocotb/build/libs/x86_64/gpivpi.vpl
ln -sf libvpi.so /home/krishna/cocotb/build/libs/x86_64/cocotb.vpi
make[2]: Leaving directory `/home/krishna/cocotb/lib/vpi'
PYTHONPATH=/home/krishna/cocotb/build/libs/x86_64:/home/krishna/cocotb:/home/krishna/cocotb/examples/functionality/tests: LD_LIBRARY_PATH=/home/krishna/cocotb/build/libs/x86_64:/usr/local/lib:/usr/local/lib/python3.4/lib-dynload:/home/krishna/cocotb/build/libs/x86_64/:/usr/local/lib:/usr/local/lib:/usr/local/lib:/usr/local/lib MODULE=test_cocotb,test_discovery,test_external,test_regression TESTCASE= TOPLEVEL=sample_module qverilog -64 -R -pli /home/krishna/cocotb/build/libs/x86_64/libvpi.so - +acc  /home/krishna/cocotb/examples/functionality/tests/../hdl/sample_module.v
[... cut ...]
# //  THIS WORK CONTAINS TRADE SECRET AND PROPRIETARY INFORMATION
# //  WHICH IS THE PROPERTY OF MENTOR GRAPHICS CORPORATION OR ITS
# //  LICENSORS AND IS SUBJECT TO LICENSE TERMS.
# //
# Loading work.sample_module(fast)
# Error loading design
Error loading design
# End time: 12:41:18 on Feb 27,2015, Elapsed time: 0:00:22
# Errors: 2, Warnings: 0
make[1]: *** [results.xml] Error 12
make[1]: Leaving directory `/home/krishna/cocotb/examples/functionality/tests'
make: *** [sim] Error 2
[krishna@krishna tests]$

themperek, 2015-02-27:
With questa (and icarus) on rhel 5 I am successfully using: http://continuum.io/downloads for python and http://linux.web.cern.ch/linux/devtoolset/Red_Hat_Developer_Toolset-2-2.0_Release_Notes-en-US.pdf

stuarthodgson, 2015-02-27:
@krishnasmtvs 

Can you post the output of

$> make clean
$> make SIM=questa

I need to see the output of the build of libcocotb.so

Your output above does not show this as cocotb.vpi is a symlink.

Stu

commit 9d3f44e7 (2015-02-26): Fixes #217: Depend on the correct variable
closed 2016-04-01 by stuarthodgson