cocotb/cocotb #218: It would be very useful to be able to access multidimensional nets

GordonMcGregor, 2015-03-02:
Is it possible to map vpiNetArray types?

I got the following warning:
 252.00ns WARNING  cocotb.gpi                   VpiImpl.cpp:106  in create_gpi_obj_from_handle      Not able to map type 114 to object.

From vpi_user.h :
#define vpiNetArray         114   /* multidimensional net */

i.e.,  verilog multidimensional arays of packed/ unpacked data, such as

  output [`BUS_WDITH-1:0] signals[`INSTANCES-1:0];

Is this something that could be added, to return a Python array of signals, or is there a better workaround?

GordonMcGregor, 2015-03-02:
I saw vpiRegArray was added, so I'm hopeful this is going to be quite similar...

GordonMcGregor, 2015-03-02:
I blundered through and added vpiNetArray alongside vpiRegArray as a supported type and that appears to be sufficient

chiggs, 2015-03-02:
Yup, there are a few types we need to add, for example `vpiPackedArrayVar` which is the cause of #216.  We really need to go through all the types and add them to GPI if we are able to support, but ideally also testing with code samples.

Are you able to submit a PR for `vpiNetArray`?

GordonMcGregor, 2015-03-02:
yup, I can submit the PR, assuming the way I described sounds reasonable. It is working

chiggs, 2015-03-02:
Sounds good, go for it.

elgorwi, 2015-03-03:
Yes, it would be very useful to improve this. 

Currently accessing vpiRegArray objects will result in warnings in GpiCommon.cpp. If you have access to a mixed language simulator you can run the 'mean' example the see the warnings.
I think changing the LOG_WARN statements to LOG_DEBUG in gpi_get_handle_by_index will get rid of the warnings.

In addition, it would be nice to extend the python side to provide better support for multi-dimensional arrays. Being able to assign a list directly to a 2D array rather than iterating and assigning the elements of the 2D array one-by-one would be more convenient. (and performance might be better too if the iteration was done in C) 
Another benefit would be that the drivers and monitors could be used with 2D data arrays as well.

```python
vals = [1,2,3,4]

# current array assignment
for idx, val in enumerate(vals):
    dut.data_bus[idx] <= val

# proposed array assignment
dut.data_bus <= vals
```

closed 2020-04-23 by ktbarrett