cocotb/cocotb #220: Add a clean way to link in CocoTb but not do anything at startup

GordonMcGregor, 2015-03-03:
I'm using cocotb in amongst a bunch of UVM and legacy verilog testbenches. The VPI layer is linked in for all the builds, along with various other VPI models. Right now, if CocoTb starts and isn't handed a TOPLEVEL in the environment it fails out the simulation. As many of the tests, and test writers have no idea what CocoTb is and aren't currently using it, it would be nice if it just disabled itself with a reasonable message, at least for these sort of mixed simulation environments.  

I realise this is heresy to say that CocoTb isn't always the main testbench ;) 

I fixed/ worked around it by having embed_sim_init return 0 as opposed to -1 when warning that it was unable to find the root instance (lib/embed/gpi_embed.c::embed_sim_init ) and print a slightly more descriptive error message (e.g., acknowledging it is actually a message from CocoTb for example)

chiggs, 2015-03-17:
@GordonMcGregor can you provide some more context about what you're doing?

In general I agree this could be a useful feature.  For example if we had some Python based protocol checkers or coverage trackers we could effectively use DPI to bind them to instances in UVM / legacy testbenches, thus allowing better re-use of the Python code.

Also following on from your recent tweet, I think it would be pretty straightforward to expose an API via DPI for embedding the Python interpreter and requesting Cocotb loads a Python module or calls a Python function.  From within that function you would have access to the `simulator` module and could therefore traverse the hierarchy etc.

What I'm less sure about is what happens next.  It's impossible to call into the simulator asynchronously from an external context but that might not be a problem.  I'm curious to know what you might want to pass between Verilog and Python and therefore what the DPI API would need to look like.

GordonMcGregor, 2015-03-18:
This is a fairly large topic.  I'll try to answer with what I've done in the past and what I'm currently doing.

Currently: 

We have a build environment and many unit level, cluster level and full chip level simulation views that are built, all using the same set of fairly complex run/regression scripts.

The various DPI and VPI hooks that we have are stored in a common place and built/ linked into every simulation.  These include svlib, some Matlab integration libraries etc.  I add CocoTB into that mix because I actually want to use Python as the verification bench for one of the unit environments (and maybe want to expand the usage to reuse the model/ checkers in the top level benches too at some point in the future)  Basically, I want CocoTB to be available in all the environments, even though most of those environments currently have no idea about CocoTB.

So I'd like it to fail quietly if it isn't used in those environments (in the same way for example our Matlab model managers might be linked in but not used, or svlib might be available, but not used)

That's the simple case, that can be addressed by failing gracefully if there is no TESTCASE environment, for example.

The second use case is a that I'd like to use Python, from systemverilog testbenches. For example, suppose I have a pure SV bench, but I want to use JSON configuration files for config data. I could write a JSON parser in SV.  I could link a C library. Or I could make a Python DPI call.  This was largely what PyHVL excelled at - exposing a Python DPI to (s)verilog cleanly.  It had a crappy build environment and many issues, though, so I'd like to be able to do the same thing with CocoTB.  I think I made this work before, by having the CocoTB simulation integration started by a verilog DPI call, that would then launch CocoTB in a DPI context, hook up all the usual VPI stuff and then allow DPI python calls back and forth between the systemverilog bench and the python bench.  That would then let me use Python things for things python is good at, within a normal systemverilog testbench (or python/ systemverilog bench)

E.g., I've embedded web servers into test benches quite easily using this model.

The third, maybe most interesting thing I'd like to be able to do with that DPI interface is make calls to UVM agents. I think the ideal world is that low level testbench features (drivers, stuff that touch RTL, things with detailed timing and obscure assertion requirements) exist in (s)verilog. The upper layer of stimulus and test generation could be in Python, and then call over DPI to drive transactions.  The best example of this I have is that we use PCIe testbench IP that we purchase. It is SV-UVM, costs quite a lot of money, is verified and I am certainly not about to write a Python PCIe bus driver (either PIPE or serial interface - that's a lot of complexity) I'd love to be able to reuse that SV_UVM PCIe agent, but have Python provide upper level stimulus (really we are driving network traffic over that PCIe BFM, so I want to use SCAPY/ wireshark etc to actually drive that interface)

I think all this needs is the CocoTB environment to be in the DPI context, then I could make calls into the agent's queue from python (with some appropriate UVM transaction translation)

krupan, 2015-04-20:
" I think the ideal world is that low level testbench features (drivers, stuff that touch RTL, things with detailed timing and obscure assertion requirements) exist in (s)verilog. The upper layer of stimulus and test generation could be in Python, and then call over DPI to drive transactions. The best example of this I have is that we use PCIe testbench IP that we purchase. It is SV-UVM, costs quite a lot of money, is verified and I am certainly not about to write a Python PCIe bus driver (either PIPE or serial interface - that's a lot of complexity) I'd love to be able to reuse that SV_UVM PCIe agent, but have Python provide upper level stimulus"

I came here to type almost exactly the same thing.  At this point in the game there is existing low-level SV/UVM verification IP that is not going to be rewritten in python, at least not until cocotb *really* proves itself.  A really easy way to prove cocotb out would be to mix it in with existing SV/UVM testbench elements in the way Gordon has described.

saturn77, 2015-04-20:
Congruent with this thread but with a slight offset, I am wondering if anyone has considered wrapping the System Verilog DPI with Cython, and then having all the System Verilog DPI functions / tasks exposed in Python as importable modules. cocotb would then be able to use this modules. 

I have actually looked at this, and have spent time reading the IEEE 1800 standard. However, it's obfuscated imho. The DPI seems powerful, but I haven't seen enough of it yet.

tsengr0916, 2018-05-17:
I have a made a piece of C code that can pass the SV class object thru DPI-C, to Python. Thise code works fine without COCOTB. I have the problem to integrate this C code into COCOTB environment. Any helps/suggestions will be highly appreciated. The C code basically receives the string arguments from SV, and then calls another python function to process the arguments:

#include <Python.h>
#include <string.h>
#include <stdio.h>
#include "dpiheader.h"
#include "vpi_user.h"

int c_call_py(int argc, const svOpenArrayHandle argv)
{
    
    PyObject *pName, *pModule, *pFunc;
    PyObject *pArgs, *pValue;
    char* c_argv[argc];
    
    for (int i=0; i<argc; i++) {
      char *str;
      str = *(char **)svGetArrElemPtr1(argv,i);
      c_argv[i] = str;
      vpi_printf("C argv[%0d]=%s\n", i, c_argv[i]);      
    }
    
    if (argc < 2) {
       fprintf(stderr,"Usage: c_call_py pythonfile funcname [args]\n");
       return 1;
    }

    Py_Initialize();

    // Error checking of pName left out
    pName = PyUnicode_FromString(c_argv[0]);
      
    pModule = PyImport_Import(pName);
    Py_DECREF(pName);

    if (pModule != NULL) {
        // pFunc is a new reference 
        pFunc = PyObject_GetAttrString(pModule, c_argv[1]);

        if (pFunc && PyCallable_Check(pFunc)) {
            pArgs = PyTuple_New(argc - 2);
            
            printf("C num args is %d\n",argc); 
                        
            for (int i = 0; i < argc - 2; i++) {
              pValue = PyUnicode_FromString(c_argv[i+2]);
              //vpi_printf("i=%0d\n", i);
              if (!pValue) {
                Py_DECREF(pArgs);
                Py_DECREF(pModule);
                fprintf(stderr, "Cannot convert argument\n");
                return 1;
              }
            
              // pValue reference stolen here:
              PyTuple_SetItem(pArgs, i, pValue);            
            }
            
            // call python function with args
            pValue = PyObject_CallObject(pFunc, pArgs);
            
            Py_DECREF(pArgs);
            if (pValue != NULL) {
                //vpi_printf("Result of call: %ld\n", PyLong_AsLong(pValue));
                Py_DECREF(pValue);
            }
            else {
                Py_DECREF(pFunc);
                Py_DECREF(pModule);
                PyErr_Print();
                fprintf(stderr,"Call failed\n");
                return 1;
            }
        }
        else {
            if (PyErr_Occurred())
                PyErr_Print();
            fprintf(stderr, "Cannot find function \"%s\"\n", c_argv[1]);
        }
        Py_XDECREF(pFunc);
        Py_DECREF(pModule);
    }
    else {
        PyErr_Print();
        fprintf(stderr, "Failed to load \"%s\"\n", c_argv[0]);
        return 1;
    }
    Py_Finalize();
    return 0;
}

I compiled and linked the above C code to c_call_py.so file. Then in my Modelsim command, I did this:
vsim -sv_root /path/to/the/so/file/dir -sv_lib -sv_lib c_call_py ... 

In my SV environment, I have this:
  import "DPI-C" context function int c_call_py(int argc, input string argv[]);

And I could build an array and to call the C code from the SV code. This works fine if COCOTB is not involved. If I call the C code from SV within the COCOTB environment, it crashes and says this:

# ** Fatal: (SIGSEGV) Bad handle or reference.
#    Time: 200 ns  Iteration: 3  Process: /env_pkg::init_seq::body/#FORK#101(#ublk#207382391#101)_7fefe43402f File: /prj/tbs/5g/sandiego/hw/main/users/hctseng/dev5g/rxf/blocks/rx_front/verif/sim/env/src/tb_top_if.sv
# ** Fatal: (vsim-4) ****** Memory allocation failure. *****
# Attempting to allocate 140737438350376 bytes

Please advise. Thanks very much,

Richard

ktbarrett, 2020-05-05:
xref #1225

ktbarrett, 2021-01-30:
Renaming, it seems like this issue is more about DPI integration than "not doing anything at start up". As I expressed in #1989, I don't think this is possible. But will keep this question open.

awmoore-intel, 2025-05-12:
Python can easily call DPI functions defined in SV using cffi. To go the other way and call into Python methods, it would require some wrapper code, but is possible. Or data could be transferred via VPI by just setting signals from the verilog testbench and using callbacks for value changes on the Python side. However, I'm not sure if there's any support in cocotb is required for any of this.

ktbarrett, 2025-05-13:
There is some additional things that need to be done. Calling DPI export functions from Python requires calling `svSetScope`. And as you stated about calling into Python methods with DPI import functions. If they are pure Python, that's no problem, albeit exceptions will not be handled gracefully. But if you need to run a Task or have Exceptions shutdown the test cleanly, you'll need to enter into the event loop and that will need a mutex lock.

cocotb could provide some simple method to for doing the former and the later will need some work but is definitely worth doing.