cocotb/cocotb #223: [CoCoTB][Running of endian swapper example]

PremTVS, 2015-03-27:
Hi,
After going through the "CoCoTB" tutorial, couple of simple examples helped in knowing the concept behind using python.
But when "endian swapper" example from installation path is run which is coded with CoCoTB framework, few errors were appeared as described in the attachment snapshots .
LD_LIBRARY_PATH is set appropriately.
Tools used:
1. Python 2.6.1
2.  irun(64): 13.10-s007

Command used:
make SIM=ius

Any solution would be helpful.

[image]

[image]

chiggs, 2015-03-27:
@PremTVS could you just sanity check that Icarus works OK?  Just to verify that the build is OK.

Also what OS are you using?

chiggs, 2015-03-27:
@PremTVS also note there is an anomaly in the traceback.  You say you're using Python 2.6.1 but the Python version in your traceback is 2.7.

Do you have multiple versions of Python installed? Is it possible that we're linking against the wrong Python version?

PremTVS, 2015-03-27:
@chiggs 
Thanks for the reply.
The operating system is "Fedora" and unfortunately I cannot run with "Icarus" as tool is not installed for use.
Coming to the python versions, 
When the example is ran first time with make command, it complained of "python2.7.so" file which it could not find.
After setting the LD_LIBRARY_PATH appropriately(where all .so files are present), I do not see any further error with respect to the above file.
Still if you suspect the issue is due to Python installation,
Please let us know whether "Python version" should be appropriately mentioned with any of the setup files inside "cocotb" installation area so that all the time we see a single Python version available.

chiggs, 2015-03-27:
You shouldn't need to change `LD_LIBRARY_PATH` to link against Python as the build system should find it and link correctly.  My guess is that you have multiple versions of Python installed which is causing some confusion?

If you just type `python -V` on a command prompt which version do you get?

Can you also uncomment the `debug` rule in `cocotb/makefiles/Makefile.pylib` and post the output of `make debug`?  You should see something like this:

```
make debug
Debug from Makefile.pylib
        Using PYTHON_VERSION=2.7
        Using PYTHON_LIBDIR=/usr/lib64
        Using PYTHON_DYN_LIB=libpython2.7.so
        Using PYTHON_DYNLIBDIR=/usr/lib64/python2.7/lib-dynload
        Using PYTHON_INCLUDEDIR=/usr/include/python2.7
file /usr/lib64/libpython2.7.so
/usr/lib64/libpython2.7.so: symbolic link to `libpython2.7.so.1.0'
file -L /usr/lib64/libpython2.7.so
/usr/lib64/libpython2.7.so: ELF 64-bit LSB shared object, x86-64, version 1 (SYSV), dynamically linked, BuildID[sha1]=2c1a6d733e01b39b1bc445bc54c0b833f8fbe270, stripped
```

PremTVS, 2015-03-27:
@chiggs 
Please find the outputs for the said commands below:

python -V -> Python 2.6.1
make debug ->
Debug from Makefile.pylib
        Using PYTHON_VERSION=2.6
        Using PYTHON_LIBDIR=
        Using PYTHON_DYN_LIB=libpython2.6.so
        Using PYTHON_DYNLIBDIR=
        Using PYTHON_INCLUDEDIR=/opt/openoffice.org3/basis-link/program/python-core-2.6.1/include/python2.6
file /libpython2.6.so
/libpython2.6.so: ERROR: cannot open `/libpython2.6.so' (No such file or directory)
file -L /libpython2.6.so
/libpython2.6.so: ERROR: cannot open `/libpython2.6.so' (No such file or directory)

chiggs, 2015-03-27:
So it looks like your path is messed up - you're pulling in python bundled with OpenOffice rather than the system version (which will almost certainly be Python 2.7 on reasonably recent versions of Fedora).  The OpenOffice version doesn't include the shared library and definitely isn't a good idea to use this, particularly as your PYTHONPATH seems to be picking stuff up from `/usr/local/lib/python2.7`.

Can you remove any superfluous paths from your `PATH` environment variable and ensure that `which python` points to  the system install of Python, or the version.

Note you'll also have to comment out the debug rule to try and build again.  Be sure to run `make clean`  to clear out the old libraries and force a full re-compile using the correct Python environment.

PremTVS, 2015-03-27:
@chiggs 
Thanks for the pointer and we will get back to you following the re-attempt of running the framework with necessary updates.

PremTVS, 2015-03-31:
@chiggs 
We sorted out the issue with python installation and now also I see same error after running the code.
Please find below:
pyhton -V -> Python 2.7.6
make SIM=ius debug ->
Debug from Makefile.pylib
        Using PYTHON_VERSION=2.7
        Using PYTHON_LIBDIR=/usr/lib64
        Using PYTHON_DYN_LIB=libpython2.7.so
        Using PYTHON_DYNLIBDIR=/usr/lib64/python2.7/lib-dynload
        Using PYTHON_INCLUDEDIR=/usr/include/python2.7
file /usr/lib64/libpython2.7.so
/usr/lib64/libpython2.7.so: symbolic link to `/usr/local/lib/libpython2.7.so'
file -L /usr/lib64/libpython2.7.so
/usr/lib64/libpython2.7.so: ELF 64-bit LSB shared object, AMD x86-64, version 1 (SYSV), not stripped

Is the above 2 outputs for the commands are correct.

themperek, 2015-03-31:
For me works  with: irun: 13.10-s024
For python I use: http://continuum.io/downloads

chiggs, 2015-03-31:
@PremTVS Yes that looks good - cocotb has found the correct Python installation and shared object files.

As mentioned above, to run the test you'll need to comment out the `debug` rule as this overrides the default rule.

PremTVS, 2015-04-01:
@chiggs 
Thanks for your help.
I freshly cloned "cocotb" and ran "endian swapper" example with updated python installation.
Tools used:
Python : Python 2.7.6
IUS : irun(64)        13.10-s007

I am able to run "endian swapper" example successfully (without any errors). As my intention is to understand the framework, I have put break points accordingly and ran it by enabling python debugger. I observed that control is flowing rapidly across the framework.
I would like to know is there any document or any alternate way (running other existing examples) to better understand the flow.
I mean to ask the description for triggering, scheduling, and regression mechanism in the flow and etc.

chiggs, 2015-04-01:
@PremTVS The internal mechanisms aren't documented but here's a very brief synopsis:

GPI is an abstraction of VPI, VHPI or FLI depending on the interfaces available from the simulator.  This is all written in C/C++.  The interaction between Python and GPI is via a Python extension module called `simulator` which provides routines for traversing the hierarchy, getting/setting and objects value, registering callbacks etc.

All callbacks run through the `scheduler.py` block which basically works out which coroutines to wake up and schedules them.  Tests are auto-discovered using Python introspection and a regression by default just runs all tests found that don't have the `skip` flag passed to the `@cocotb.test()` decorator.

It's probably best not to get too hung up on the internals - we deliberately hide and abstracting the complicated internals to provide convenient usage at the high-level.  An understanding of coroutines is really all that's required.

The easiest way to get up to speed is probably by examining some of the examples.  There are a few others that aren't in the repository:

* http://www.edaplayground.com/s/example/106
* https://github.com/chiggs/oc_jpegencode

themperek, 2019-10-16:
Thank you for reporting and discussion. It looks like an obsolete issue. Please reopen if you still have problems. 

@cmarqu Thank you for pointing out.

closed 2019-10-16 by themperek