cocotb/cocotb #226: Python _imp__*

reedjosh, 2015-04-04:
Hey guys,

This this is likely just my lack of knowledge. I've been trying to get cocotb running for use on Windows, and I've not had any luck. I'm an intern, and this is one of my tasks. 

I've got mingw, python, and iverilog installed. When I run the makefile in the tests folder I get several issues of the form:

io_module.o:io.c:(.text+0x10e): undefined reference to '_imp__PyErr_Occured'

I've installed swig as well, but I may have done so incorrectly. I've also tried making the distutils.cfg file and libpython.a file, but with no luck.

reedjosh, 2015-04-04:
[image]

stuarthodgson, 2015-04-04:
Hi,

Could you just confirm that you are running version 1.0.0?

Stu

reedjosh, 2015-04-04:
[image]

reedjosh, 2015-04-04:
Is 1.0.0 different than 1.0?

potentialventures, 2015-04-04:
No it's not. I'm away from home at the moment so won't be able to look at
this until Monday afternoon GMT. It's not an error I've seen and im more
concerned about the error from ld rather than the symbols.

Can you try the functionally test by rubbing make in
examples/functionality/tests please. The error you are seeing is in
building a lib for a test case not a cocotb library. So may not impact you
using the framework.

Stu

reedjosh, 2015-04-04:
[image]

reedjosh, 2015-04-04:
Here you are sir. Thanks for the quick support. I'm pretty novice with a lot of these tools, so please don't hesitate to tell me if I do something stupid.

jeremyherbert, 2015-04-05:
Hi reedjosh,

Could you post your Python version? python --version is the command you
want.

reedjosh, 2015-04-05:
[image]

jeremyherbert, 2015-04-05:
Ok, how about this one?

python -c "import platform; print platform.architecture()"

reedjosh, 2015-04-05:
[image]

elgorwi, 2015-04-06:
Ok, note the endian-swapper HAL cosimulation example currently doesn't run under windows. 
However, the non-cosimulation related tests of the endian swapper example should work fine under windows. 
Could you try copying the cocotb dir from /home/josh/cocotb to c:/cocotb and 'make clean' and then 'make' again?
What output do you get when running 'which gcc' and 'which python'?
Also, which python distribution did you install?

themperek, 2019-10-16:
#1099

themperek, 2020-04-16:
In the current development version, windows support should be better now.
Please also check https://docs.cocotb.org/ for instructions.

Since this issue is from 2015 I am closing but feel free to reopen in case the issue is still there.

closed 2020-04-16 by themperek