cocotb/cocotb #227: Support for verilator

mrvkino, 2015-04-05:
Do you think you will add support for verilator in the future? http://www.veripool.org/wiki/verilator

Is there a way for me to add it (easily) in the meantime?

jeremyherbert, 2015-09-03:
I have started to become quite interested in this too. Has anyone taken a good look at this yet ?

Martoni, 2016-03-16:
I'm also interested by it, but I don't know where to begin.

Dolu1990, 2016-06-23:
:+1:   I'm also very interrested

juliusbaxter, 2018-08-27:
Has anyone given this a go? Do we think it's in the realm of the sort of work that could be done by a GSoC student? If so we could float it for next year's programme.

jeremyherbert, 2018-08-27:
I seem to recall that the majority of the work for this is on the verilator end, not the cocotb one. And the maintainer of verilator did not seem all that keen on VPI at the time, because he believed non-compiled test benches were of limited use.

juliusbaxter, 2018-08-27:
Interesting! Thanks for the update. I shall be speaking to the maintainer of Verilator at ORConf in less than a month so I'll be sure to try and persuade him as to the value of non-compiled test benches.

Dolu1990, 2018-08-27:
I'm kind of in the same situation with Verilator used as simulator by the SpinalHDL simulation API.
It had require a little extra layer on the top of verilator to emulate simulation time step and sentitivity.
https://github.com/SpinalHDL/SpinalHDL/blob/master/sim/src/main/scala/spinal/sim/SimManager.scala#L92  
to line 142

cmarqu, 2019-01-20:
As said in #320, @Nic30 has some Verilator binding with Python in https://github.com/Nic30/pycocotb FWIW

pwiecha, 2019-05-23:
Basic Verilator infrastructure #943

imphil, 2019-05-23:
Just for all people following this bug: The cocotb part is the easy one, more work is required on the Verilator side. Initial Verilator patches are currently being prepared, I'm expecting an initial release in a couple of weeks.

Nic30, 2019-05-23:
Happy to hear it, I do have Verilator working with the cocotb, however I am not able to finish this thing in the next few weeks because it is not part of my PhD thesis and I am not getting paid for this. And things related to my PhD are getting more urgent every day. If I found any OpenStack/Kubernetes/Openflow/OVS expert which could answer my questions I can can help on this.

kgugala, 2019-06-19:
We have integrated Verilator with cocotb and the code is available here: https://github.com/chipsalliance/cocotb/tree/verilator-support

The integration requires changes in both Verilator and Cocotb. The easiest way to test it is to follow the guide in the readme in https://github.com/chipsalliance/tools-cocotb-verilator-build

We'd like to push all the changes upstream. Any comments are appreciated.

toddstrader, 2019-06-19:
As mentioned before, the major part of this is the Verilator piece.  I think it would be more prudent to land that first.  Also, this conversation has become somewhat disjoint.  I have a couple follow-up comments/questions here from a little while ago:
https://github.com/wallento/verilator/issues/1

I'm not sure where the best place is to get everyone's attention on this issue.  Any suggestions on where we should rally (this issue, @wallento's PR, @wallento's forks or @kgugala's meta-repo)?

More specifically, I think we need to first agree what is missing/broken in terms of Verilator VPI support.  Then we should land those changes before trying to tie cocotb to it.  One thing I am definitely worried about is subtly telling Verilator that its being run under cocotb and then having it alter its VPI behavior based on that.  This is probably not going to fly.
https://github.com/chipsalliance/cocotb/blob/57a00bdd71a3207dbc6a42d5b9b9d58d8981a818/cocotb/share/lib/verilator/verilator.cpp.in#L27

imphil, 2019-06-19:
The general rule is: enhance Verilator to get sufficiently complete, standards-compliant VPI support. If that's not possible (i.e. a patch is rejected by Wilson for good reasons), then we can think about adding workarounds. VPI is very well defined by the standard and by multiple other implementations, tool-specific workarounds must remain the last resort.

So the only path forward is:
- Get all necessary changes into Verilator
- Then do a PR here to add support for cocotb. It most likely can build on top of this PR.

mgielda, 2019-06-19:
Yes, I agree. Perhaps @kgugala did not mention it explicitly, but that is the plan - verilator vpi first. I just asked him to link back to the stuff we worked on so that the conversations can join again :) AFAIK we will be doing prs against Verilator until end of this week.

wallento, 2019-06-19:
Yeah, in particular there is one topic only covered by workarounds so far: https://github.com/wallento/verilator/issues/5

Or did I miss something. I think that will be a bit tricky and will need some rounds of review, but I thinks it desirable to land systemtf support without workarounds.

morsingnet, 2019-06-19:
Is there an example that we can follow to test this ?
Im very interested since i use Verilator for linting right now.
I have cocotb running the "littleEndian" example and Icarus simulator and wanted to see if it would run with Verilator instead.

wallento, 2019-06-19:
Following @toddstrader's suggestion, I propose we move the discussion here: https://gitter.im/librecores/cocotb-verilator

wallento, 2019-06-19:
That seems like the easiest form of communication to get everyone aligned on whats going on and coordinate in getting the best code ready for upstream

themperek, 2020-04-16:
Basic Verilator support is now available in cocotb. #943 

Since this issue is from 2015 I will close. Please feel free to report in case you find issues.

closed 2020-04-16 by themperek