cocotb/cocotb #231: VCS segfaults when compiling code with overridden system calls

chiggs, 2015-04-10:
In the `sim_exit` testcase:

```
Parsing design file '/home/chiggs/code/github/cocotb/examples/sim_exit/tests/../hdl/close_module.v'
Top Level Modules:
       close_module
TimeScale is 1 ps / 1 ps
Starting vcs inline pass...
Got SIGSEGV

An unexpected termination has occurred in /tools/synopsys/vcs/I-2014.03-2/amd64/bin/vcs1 because of a fatal signal

Stmt $info at /home/chiggs/code/github/cocotb/examples/sim_exit/tests/../hdl/close_module.v, 58
Stmt  at /home/chiggs/code/github/cocotb/examples/sim_exit/tests/../hdl/close_module.v, 58
Stmt  at /home/chiggs/code/github/cocotb/examples/sim_exit/tests/../hdl/close_module.v, 55
Process  /home/chiggs/code/github/cocotb/examples/sim_exit/tests/../hdl/close_module.v, 55

During addModuleItems Module close_module declared at /home/chiggs/code/github/cocotb/examples/sim_exit/tests/../hdl/close_module.v, 33
 XModule close_module declared at /home/chiggs/code/github/cocotb/examples/sim_exit/tests/../hdl/close_module.v, 33
 Hsim mod number=1 

Command line: /tools/synopsys/vcs/I-2014.03-2/amd64/bin/vcs1 -Mcc=gcc -Mcplusplus=g++ -Masflags= "-Mcfl= -pipe -fPIC -O -I/tools/synopsys/vcs/I-2014.03-2/include " "-Mldflags= " -Mout=simv -Mamsrun= -Mvcsaceobjs= "-Mobjects= /tools/synopsys/vcs/I-2014.03-2/amd64/lib/libvirsim.so /tools/synopsys/vcs/I-2014.03-2/amd64/lib/liberrorinf.so /tools/synopsys/vcs/I-2014.03-2/amd64/lib/libsnpsmalloc.so " -Msaverestoreobj=/tools/synopsys/vcs/I-2014.03-2/amd64/lib/vcs_save_restore_new.o -Mcrt0= -Mcrtn= -Mcsrc= "-Msyslibs=-ldl -lm -lm " +acc+1 -P pli.tab +define+COCOTB_SIM=1 -Xcbug=0x1 -load libvpi.so -full64 -picarchive +cli+4 +vcsd +itf+/tools/synopsys/vcs/I-2014.03-2/amd64/lib/vcsd.tab +vpi -debug=3 +memcbk +vpi -sverilog +vpi -gen_obj /home/chiggs/code/github/cocotb/examples/sim_exit/tests/../hdl/close_module.v

--- Stack trace follows:
[Thread debugging using libthread_db enabled]
0x00000033ee2ac61e in waitpid () from /lib64/libc.so.6
#0  0x00000033ee2ac61e in waitpid () from /lib64/libc.so.6
#1  0x00000033ee23e609 in do_system () from /lib64/libc.so.6
#2  0x00000000040baac5 in tracker::StackAnnotator::getGdbOutput(char const*, char const*, char const*, char const*) ()
#3  0x00000000040bb08b in tracker::StackAnnotator::dumpStackByGdb(char const*, char const*) ()
#4  0x00000000040bc3b6 in tracker::StackAnnotator::dumpProcessStack(unsigned int, char const*, char const*) ()
#5  0x00000000040b17d8 in ?? ()
#6  0x0000000003d45887 in stackTraceHandler ()
#7  0x0000000003d4420e in SigHandler ()
#8  <signal handler called>
#9  0x00000000030e8600 in SetupMBDaiCapTsk ()
#10 0x0000000002cd0e9a in vcsAddModuleSysTask(Stmt const*, mb*) ()
#11 0x0000000002cdd963 in ?? ()
#12 0x0000000002cdbd80 in ?? ()
#13 0x0000000002cdb6d5 in ?? ()
#14 0x0000000002cde653 in ?? ()
#15 0x0000000002cdfd66 in vcsPutTokenModule(XModule const*) ()
#16 0x0000000002cedebc in vcsAddModuleItems(XModule*) ()
#17 0x0000000002cf3650 in ?? ()
#18 0x0000000002cf405b in vcsCreateModuleStruct(int) ()
#19 0x00000000012320b2 in doRadify_vir2vcsAll ()
#20 0x0000000002fc21dc in ?? ()
#21 0x0000000002fc84be in c_main ()
#22 0x0000000000e68411 in main ()
Completed context dump phase data location 'Stmt'* 'Stmt'* 'Stmt'* 'Process'* 'Module'* 'XModule'*
CPU time: .646 seconds to compile
make[1]: *** [sim_build/simv] Error 1
make[1]: Leaving directory `/home/chiggs/code/github/cocotb/examples/sim_exit/tests'
make: *** [sim] Error 2
```

Commenting out the system calls (`$info`, `$fatal`) etc. allow compilation to continue.

chiggs, 2015-04-10:
For the time being I will disable this regression if targeting VCS.

hashhsah, 2019-12-22:
VCS has a `-user_vpiobj` option for user to specify the .c file that contains the definition of the  `vpi_register_systf` function.

In addition, the current `Makefile.vcs` does not allow mixed language simulation, while VCS does support it. One could use `vlogan` and `vhdlan` to compile verilog and vhdl source files, respectively in the analysis phase into a work library, then use `vcs` to elaborate (link) from the library to get the `simv` executable.

cmarqu, 2019-12-22:
@hashhsah I have been working on this a bit in #1103, but got stuck because of the different vhpi_user.h file. This should be solvable with some effort, any help appreciated.

garmin-mjames, 2020-10-26:
The system task overload were removed in #2133. Can this be closed now?

closed 2020-10-26 by ktbarrett

cmarqu, 2020-10-26:
This test itself does not exist anymore, and its replacement ``test_fatal`` runs fine in VCS (with #2163 applied).