cocotb/cocotb #232: Tidy up selection of implementation layer based on language in use

stuarthodgson, 2015-04-10:
Some Makefiles use TOPLEVEL_LANG=<lang>, some use GPI_IMPL=<layer>

We need to unify this across the inbuilt tests and make sure the documentation is consistent. 

The scheme was supposed to be

User specifies TOPLEVEL_LANG, internally we pick the correct layer for that lang based on simulator in use. If the user also specifies GPI_EXTRA then extra implementation libs are loaded for mixed language support.

However too much of this seems to be duplicated in the test makefiles and some should probably be pushed down into the sim makefiles.

Then we can document what env vars a user writing their own makefiles needs to set in order to get the behaviour they want to run a test.

Suggestions welcome

leftink, 2016-01-06:
@stuarthodgson What are your thoughts on removing the need for the user setting GPI_EXTRA?

The design makefile already sets the VHDL_SOURCES and/or the VERILOG_SOURCES.  Using these two in conjunction with TOPLEVEL_LANG we have all the information we need to load the correct Library and set the GPI_EXTRA environment variable.

For example:
  if TOPLEVEL_LANG==vhdl:
    "set the sim args to load vhpi/fli"
    ifneq ($(VERILOG_SOURCES),)
      GPI_EXTRA = vpi
    endif
  endif
  if TOPLEVEL_LANG==verilog:
    "set the sim args to load vpi"
    ifneq ($(VHDL_SOURCES),)
      GPI_EXTRA = vpi
    endif
  endif
 
If we make these changes, "exporting" of the GPI_EXTRA would have to be move to Makefile.sim so the variable is set before exporting.

stuarthodgson, 2016-01-09:
@leftink , yes this is what I have been thinking as well. I'll try and get around to this soon as the untidyness has been frustrating for a while.

closed 2016-04-01 by stuarthodgson