cocotb/cocotb #233: Exit codes from default `make` invocation

chiggs, 2015-04-10:
The way exit codes are reported and caught has evolved over time.

Originally, the exit code was only non-zero if there was an uncaught error or compilation failure.  Failing tests did not give a non-zero exit code, the user was expected to parse the `results.xml` files.  This behaviour was driven by the way Jenkins reporting is done.

This clearly makes it too easy to miss failures when running from the command line and doesn't integrate well with Travis or other tools.  The `report_results.py` script was added to change this behaviour and make the exit code non-zero if any tests fail.

`report_results.py` introduces a dependency on `xunitparser` which doesn't appear to work on Python 2.6 when installed from pip:

```
./bin/report_results.py combined_results.xml
Traceback (most recent call last):
  File "./bin/report_results.py", line 34, in <module>
    exit(report_results(sys.argv[1]))
  File "./bin/report_results.py", line 20, in report_results
    ts, tr = xunitparser.parse(open(xml_filename))
  File "/usr/lib/python2.6/site-packages/xunitparser.py", line 217, in parse
    return Parser().parse(source)
  File "/usr/lib/python2.6/site-packages/xunitparser.py", line 141, in parse
    return self.parse_root(root)
  File "/usr/lib/python2.6/site-packages/xunitparser.py", line 151, in parse_root
    tr = ts.run(self.TR_CLASS())
  File "/usr/lib64/python2.6/unittest.py", line 459, in run
    test(result)
  File "/usr/lib64/python2.6/unittest.py", line 299, in __call__
    return self.run(*args, **kwds)
  File "/usr/lib/python2.6/site-packages/xunitparser.py", line 59, in run
    tr.addSkip(self, '%s: %s' % (self.typename, self._textMessage()))
``

So I would like to remove this dependency.  I propose changing `combine_results.py` to return a non-zero exit code if there are any failures.  We'll remove `report_results.py` completely.

I'll also add another top-level rule called `jenkins` which allows the old mechanism of saving the return code for build errors rather than failed tests.

stuarthodgson, 2015-04-10:
Looks sensible

eric-wieser, 2020-04-09:
Looks like this was closed in #234

closed 2020-04-09 by eric-wieser