cocotb/cocotb #241: compilation errors

bhaskar21, 2015-04-22:
Hi All,

When we are trying to compile cocotb am getting below errors:

 make
/prj/qct/sysadmin/sw_installs/bhaskar/cocotb/new2/cocotb/examples/endian_swapper/tests/../../../makefiles/Makefile.inc:49: Extraneous text after `else' directive
/prj/qct/sysadmin/sw_installs/bhaskar/cocotb/new2/cocotb/examples/endian_swapper/tests/../../../makefiles/Makefile.inc:54: /makefiles/Makefile.paths: No such file or directory
/prj/qct/sysadmin/sw_installs/bhaskar/cocotb/new2/cocotb/examples/endian_swapper/tests/../../../makefiles/Makefile.inc:55: /makefiles/Makefile.pylib: No such file or directory
/prj/qct/sysadmin/sw_installs/bhaskar/cocotb/new2/cocotb/examples/endian_swapper/tests/../../../makefiles/Makefile.inc:73: Extraneous text after `else' directive
/prj/qct/sysadmin/sw_installs/bhaskar/cocotb/new2/cocotb/examples/endian_swapper/tests/../../../makefiles/Makefile.inc:78: Extraneous text after `else' directive
/prj/qct/sysadmin/sw_installs/bhaskar/cocotb/new2/cocotb/examples/endian_swapper/tests/../../../makefiles/Makefile.inc:78: *** only one `else' per conditional.  Stop.

Am in SUSE10.4 host. Am using Python 2.7.9 64bit for this installation. Any help will be very useful to us. Thanks.

blorgon9000, 2015-04-23:
What version of make are you running (make -v)?

I've tested it with make 3.81.

bhaskar21, 2015-04-23:
Hi,

Thanks for looking into the issue. I was using make 3.80. I tried with 3.81 version and am getting into below error:

TestError)
     0.03ns ERROR    cocotb.regression                         regression.py:159  in tear_down                       Failed 33 out of 33 tests (0 skipped)
     0.03ns INFO     cocotb.regression                         regression.py:168  in tear_down                       Shutting down...
make[3]: Leaving directory 
.
.
.
./bin/combine_results.py
FAILURE: test_adder.adder_basic_test
FAILURE: test_adder.adder_randomised_test
FAILURE: test_endian_swapper.run_test_001
FAILURE: test_endian_swapper.run_test_002
FAILURE: test_endian_swapper.run_test_003
FAILURE: test_endian_swapper.run_test_004
FAILURE: test_endian_swapper.run_test_005
FAILURE: test_endian_swapper.run_test_006
FAILURE: test_endian_swapper.run_test_007
FAILURE: test_endian_swapper.run_test_008
FAILURE: test_endian_swapper.run_test_009
FAILURE: test_endian_swapper.run_test_010
FAILURE: test_endian_swapper.run_test_011
FAILURE: test_endian_swapper.run_test_012
FAILURE: test_endian_swapper.run_test_013
FAILURE: test_endian_swapper.run_test_014
FAILURE: test_endian_swapper.run_test_015
FAILURE: test_endian_swapper.run_test_016
FAILURE: test_endian_swapper.run_test_017
FAILURE: test_endian_swapper.run_test_018
FAILURE: test_endian_swapper.run_test_019
FAILURE: test_endian_swapper.run_test_020
FAILURE: test_endian_swapper.run_test_021
FAILURE: test_endian_swapper.run_test_022
FAILURE: test_endian_swapper.run_test_023
FAILURE: test_endian_swapper.run_test_024
FAILURE: test_endian_swapper.run_test_025
FAILURE: test_endian_swapper.run_test_026
FAILURE: test_endian_swapper.run_test_027
FAILURE: test_endian_swapper.run_test_028
FAILURE: test_endian_swapper.run_test_029
FAILURE: test_endian_swapper.run_test_030
FAILURE: test_endian_swapper.run_test_031
FAILURE: test_endian_swapper.run_test_032
FAILURE: test_endian_swapper.wavedrom_test
FAILURE: test_cocotb.test_afterdelay_in_readonly
FAILURE: test_cocotb.test_afterdelay_in_readonly_valid
FAILURE: test_cocotb.test_anternal_clock
FAILURE: test_cocotb.test_coroutine_close_down
FAILURE: test_cocotb.test_coroutine_kill
FAILURE: test_cocotb.test_edge_count
FAILURE: test_cocotb.test_either_edge
FAILURE: test_cocotb.test_falling_edge
FAILURE: test_cocotb.test_fork_and_monitor
FAILURE: test_cocotb.test_function_reentrant_clock
FAILURE: test_cocotb.test_rising_edge
FAILURE: test_cocotb.test_yield_list
FAILURE: test_discovery.access_signal
FAILURE: test_external.test_ext_call_nreturn
FAILURE: test_external.test_ext_call_return
FAILURE: test_external.test_external_from_readonly
FAILURE: test_external.test_multiple_externals
FAILURE: test_regression.issue_120_scheduling
FAILURE: test_closedown.test_failure_from_system_task
make: *** [test] Error 1

But "make install" is going well without any issues.

> make install
bin/create_files.py /local/sw_installs/bhaskar/cocotb/3.81/coc/cocotb-1.0

Installed to /local/sw_installs/bhaskar/cocotb/3.81/coc/cocotb-1.0
To uninstall run /local/sw_installs/bhaskar/cocotb/3.81/coc/cocotb-1.0/bin/cocotb_uninstall

Below is the content of the destination directory:

> ls cocotb-1.0/
bin  cocotb  include  lib  makefiles  version
> ls cocotb-1.0/bin/
cocotb_uninstall
> ls cocotb-1.0/cocotb/
ANSI.py       binary.py   clock.py        drivers     log.py       regression.py   scheduler.py    triggers.py   wavedrom.py
ANSI.pyc      binary.pyc  clock.pyc       generators  log.pyc      regression.pyc  scheduler.pyc   triggers.pyc  wavedrom.pyc
__init__.py   bus.py      decorators.py   handle.py   memdebug.py  result.py       scoreboard.py   utils.py      xunit_reporter.py
__init__.pyc  bus.pyc     decorators.pyc  handle.pyc  monitors     result.pyc      scoreboard.pyc  utils.pyc     xunit_reporter.pyc

Am new to this installation. Please let us know if we can ignore the make compilation errors.

Thanks,
Bhaskar

blorgon9000, 2015-04-24:
Should probably check why all the tests are failing. Is iverilog running?

I'd suggest running:
  examples/adder/tests
  examples/endian_swapper/tests
and looking at the results

bhaskar21, 2015-05-05:
Thanks a lot. Am able to compile cocotb without any issues now.

closed 2015-05-06 by chiggs