cocotb/cocotb #244: Unable to open lib libpython3.4.so

Snotzer, 2015-05-11:
Hi,

I've just installed cocotb (11th May) on a ubuntu machine and I'm having trouble running the example testcase e.g. endian_swapper with Icarus (0.9.7). Im using v3.4 of python, make v3.81. Do I need to declare some additional environment variable with python v3.4? An earlier cocotb was working when I had python 2.7

Here is the error from the `make` command
> vvp -M /home/damian/Projects/Work/NXP/cocotb/build/libs/i686 -m gpivpi sim_build/sim.vvp   
Unable to open lib libpython3.4.so (libpython3.4.so: cannot open shared object file: No such file or directory)
>Failed to find python lib
>     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:47   in >gpi_print_registered_impl       VPI registered
>Traceback (most recent call last):
>  File "/home/damian/Projects/Work/NXP/cocotb/cocotb/__init__.py", line 41, in <module>
>    import cocotb.handle
>  File "/home/damian/Projects/Work/NXP/cocotb/cocotb/handle.py", line 46, in <module>
>   import simulator
>ImportError: /home/damian/Projects/Work/NXP/cocotb/build/libs/i686/simulator.so: undefined symbol: PyModule_GetState
>Failed to load "cocotb"

Great project any help appreciated.

stuarthodgson, 2015-05-12:
@Snotzer 

Cocotb is unable to find the python 3.4 shared library at runtime. Is the python installation from the distribution you are using or a custom one. If custom then you need to check that there is a libpython3.4.so in your install directory. And If a distro install you can also check there is a library in  /usr/lib/python3.4.

jeremyherbert, 2015-05-12:
Hi,

I use cocotb on Ubuntu as well. For some reason unbeknownst to me, on ubuntu the library is called "python3.4m.so". This is clear if you run `python3-config --ldflags`:

```
$ python3-config --ldflags
-L/usr/lib/python3.4/config-3.4m-x86_64-linux-gnu -L/usr/lib -lpython3.4m -lpthread -ldl  -lutil -lm  -Xlinker -export-dynamic -Wl,-O1 -Wl,-Bsymbolic-functions
```

In my Ubuntu 15.04 install this seems to be corrected, but it might still be a problem for 14.04 LTS if I recall. The simple solution is just to find the directory containing the library, and running

```
sudo ln libpython3.4m.so libpython3.4.so
```

For me, that directory was listed above in the `python3-config` output: `/usr/lib/python3.4/config-3.4m-x86_64-linux-gnu`

Hopefully this fixes your problem.

Snotzer, 2015-05-12:
Actually I have a custom install from anaconda, I just installed it last week... but maybe it's not following standard practices. The libraries are here
  * ~/anaconda3/lib/libpython3.so
  * ~/anaconda3/lib/libpython3.4m.so

I created a link from to the second one and called it libpython3.4.so
I also added ~/anaconda3/lib to the PYTHONPATH

But now I get the error
```
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ImportError: No module named 'encodings'
```

I get the same if I try 
```
$ python3-config --ldflags
```
Maybe I should use a different python 3.4? I used anaconda because it has iPython integrated...

Snotzer, 2015-05-12:
In fact python is no longer correctly installed on my machine Ive obviously messed something up...

jeremyherbert, 2015-05-12:
```
$ sudo apt-get install python3-pip
$ sudo pip3 install ipython[all]
```

Will get it working using the system python.

themperek, 2015-05-12:
With my anaconda install the only think I have to do is set:
```bash
export PYTHONHOME=$(shell python -c "from distutils.sysconfig import get_config_var; print(get_config_var('prefix'))")
```

Snotzer, 2015-05-12:
Thanks for your help. I did a re-install of python 3, but whenever I export the PYTHONHOME variable it seems to break python and I need to open a new terminal. Must be something weird with my installation

Open a fresh terminal
```
~$ echo $PYTHONHOME

~$ python
Python 3.4.3 |Anaconda 2.2.0 (32-bit)| (default, Mar  6 2015, 12:04:30) 
[GCC 4.4.7 20120313 (Red Hat 4.4.7-1)] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> 
~$ export PYTHONHOME=/home/${USER}/anaconda3/bin/python
~$ echo $PYTHONHOME
/home/damian/anaconda3/bin/python
~$ python
Fatal Python error: Py_Initialize: Unable to get the locale encoding
ImportError: No module named 'encodings'
Aborted (core dumped)
~$ 
```

Snotzer, 2015-05-12:
I get the same result for any value of PYTHONHOME including empty string or the command from themperek.

chiggs, 2015-05-12:
Have you tried setting both `PYTHONPATH` and `PYTHONHOME`?

There are a [few suggestions](http://unix.stackexchange.com/a/139869/82511) that this might do the trick.

themperek, 2015-05-12:
@Snotzer PYTHONHOME in your case should be:
```
 /home/damian/anaconda3
```

Snotzer, 2015-05-12:
Thanks, I tried setting both variables but as soon as I set PYTHONHOME, the python installation is no longer working. It works again when I do unset PYTHONHOME.

Is the this variable necessary for cocotb?
I compiled without and got this, I guess it's because PYTHONHOME is not set to the cocotb directory

```
        vvp -M /home/damian/cocotb/build/libs/i686 -m gpivpi sim_build/sim.vvp   
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:47   in gpi_print_registered_impl       VPI registered
Traceback (most recent call last):
  File "/home/damian/cocotb/cocotb/__init__.py", line 41, in <module>
    import cocotb.handle
  File "/home/damian/cocotb/cocotb/handle.py", line 46, in <module>
    import simulator
ImportError: dynamic module does not define init function (initsimulator)
Failed to load "cocotb"
make[1]: Leaving directory `/home/damian/cocotb/examples/endian_swapper/tests'
```

Snotzer, 2015-05-15:
Actually there is a warning at the start of the make command which I had over looked
```
gcc -m32  -fPIC -pthread -fno-strict-aliasing -O2 -pipe -Wall -Wp,-D_FORTIFY_SOURCE=2 \
	-fexceptions --param=ssp-buffer-size=4 -D_GNU_SOURCE \
	-fwrapv -DNDEBUG -I/home/damian/anaconda3/include/python3.4m -c io.c -o io_module.o
io.c: In function ‘initio_module’:
io.c:120:5: warning: implicit declaration of function ‘Py_InitModule’ [-Wimplicit-function-declaration]
     io_module = Py_InitModule("io_module", io_module_methods);
     ^
io.c:120:15: warning: assignment makes pointer from integer without a cast [enabled by default]
     io_module = Py_InitModule("io_module", io_module_methods);
               ^
io.c:125:1: warning: control reaches end of non-void function [-Wreturn-type]
 }
 ^
```
According to stackoverflow [Py_InitModule is deprecated in py3+] (http://stackoverflow.com/questions/28305731/compiler-cant-find-py-initmodule-is-it-deprecated-and-if-so-what-should-i)
So I think this file ```endian_swapper/cosim/io.c``` needs to be modified for the case when python version is 3+

endian_swapper is the only example with an io.c, if I try another example e.g. 'adder' it works. Without any PYTHON environment variables being set.

jeremyherbert, 2015-05-15:
Endian swapper seems to work fine under python3 here: https://travis-ci.org/potentialventures/cocotb/jobs/62147748

Is it possible that you are loading the wrong version of python in your compile or run step? The error `ImportError: dynamic module does not define init function (initsimulator)` would also indicate that.

Snotzer, 2015-05-16:
That's it. I see from your link that you have the same warning but like you said it's not the source of the problem. The ```python-config``` I had was v2.7 and my python is v3.4, I fixed this by adding this link
```
     ln -s /home/damian/anaconda3/bin/python3.4m-config /home/damian/anaconda3/bin/python-config
```
The endian swapper works now. 
Thanks for your help :)

stuarthodgson, 2016-04-01:
This potential issue will have been resolved by the recent changes in #419

closed 2016-04-01 by stuarthodgson