cocotb/cocotb #246: Mixed langauage example doesn't run under ius

luked80, 2015-05-26:
I'm trying to run the mixed language example with ius, using the verilog top level.
My invocation is "make SIM=ius" (with the changes in my latest pull request).

It looks like getting the handle for the vhdl instance doesn't work correctly.

Here's the simulator output:

```shell
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:47   in gpi_print_registered_impl       VPI registered
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:47   in gpi_print_registered_impl       VHPI registered
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:229  in embed_sim_init                  Running on ncsim(64) version 10.20-s110
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:230  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
     0.00ns INFO     cocotb.gpi                                  __init__.py:103  in _initialise_testbench           Running tests with Cocotb vINCISIV102-s040 from /home/luke/cocotb_lukes
     0.00ns INFO     cocotb.gpi                                  __init__.py:119  in _initialise_testbench           Seeding Python random module with 1432608815
     0.00ns INFO     cocotb.regression                         regression.py:153  in initialise                      Found test test_mixed_language.mixed_language_test
     0.00ns INFO     cocotb.regression                         regression.py:254  in execute                         Running test 1/1: mixed_language_test
     0.00ns INFO     ..ine.mixed_language_test.0xc827750       decorators.py:186  in send                            Starting test: "mixed_language_test"
                                                                                                                               Description: 
                                                                                                                                   Try accessing handles and setting values in a mixed language environment
                                                                                                                                   
ncsim> source /tools/cadence/INCISIV102-s040/Linux/tools/inca/files/ncsimrc
ncsim> run
     0.10ns INFO     cocotb.endian_swapper_mixed        .._mixed_language.py:14   in mixed_language_test             Got: 'endian_swapper_mixed.i_swapper_sv'
     0.10ns WARNING  cocotb.gpi                                  VpiImpl.cpp:105  in create_gpi_obj_from_handle      Not able to map type 1024 to object.
     0.10ns ERROR    cocotb.gpi                                  VpiImpl.cpp:128  in native_check_create             Unable to query fetch object endian_swapper_mixed.i_swapper_vhdl
     0.10ns INFO     cocotb.endian_swapper_mixed        .._mixed_language.py:17   in mixed_language_test             Got: 'endian_swapper_mixed.i_swapper_vhdl'
     0.20ns WARNING  cocotb.gpi                                  VpiImpl.cpp:105  in create_gpi_obj_from_handle      Not able to map type 1079 to object.
     0.20ns ERROR    cocotb.gpi                                  VpiImpl.cpp:128  in native_check_create             Unable to query fetch object endian_swapper_mixed.i_swapper_vhdl.reset_n
ERROR:          VHPI       NULLPTR
         NULL value buffer passed to vhpi_get_value().
        /home/luke/cocotb_lukes/examples/endian_swapper/hdl/endian_swapper.vhdl, 52: 
ERROR:          VHPI       NULLPTR
         NULL value buffer passed to vhpi_get_value().
        /home/luke/cocotb_lukes/examples/endian_swapper/hdl/endian_swapper.vhdl, 52: 
     0.30ns INFO     cocotb.gpi                                   VhpiImpl.h:270  in get_signal_value_binstr         VHPI Error level 3: NULL value buffer passed to vhpi_get_value().
                                                                                                                               FILE /home/luke/cocotb_lukes/examples/endian_swapper/hdl/endian_swapper.vhdl:52
     0.30ns ERROR    ..ine.mixed_language_test.0xc827750           result.py:51   in raise_error                     Send raised exception: invalid literal for int() with base 2: ''
                                                                                                                                 File "/home/luke/cocotb_lukes/cocotb/decorators.py", line 191, in send
                                                                                                                                   return self._coro.send(value)
                                                                                                                                 File "/home/luke/cocotb_lukes/examples/mixed_language/tests/test_mixed_language.py", line 25, in mixed_language_test
                                                                                                                                   if int(verilog.reset_n) == int(vhdl.reset_n):
                                                                                                                                 File "/home/luke/cocotb_lukes/cocotb/handle.py", line 255, in __int__
                                                                                                                                   return int(self.value)
                                                                                                                                 File "/home/luke/cocotb_lukes/cocotb/binary.py", line 371, in __int__
                                                                                                                                   return self.integer
                                                                                                                                 File "/home/luke/cocotb_lukes/cocotb/binary.py", line 241, in get_value
                                                                                                                                   return self._convert_from[self.binaryRepresentation](self._str)
                                                                                                                                 File "/home/luke/cocotb_lukes/cocotb/binary.py", line 148, in _convert_from_unsigned
                                                                                                                                   return int(resolve(x), 2)
                                                                                                                               
     0.30ns ERROR    cocotb.regression                         regression.py:238  in handle_result                   Test Failed: mixed_language_test (result was TestError)
     0.30ns ERROR    cocotb.regression                         regression.py:159  in tear_down                       Failed 1 out of 1 tests (0 skipped)
     0.30ns INFO     cocotb.regression                         regression.py:168  in tear_down                       Shutting down...
Simulation complete via $finish(1) at time 301 NS + 0

```

imphil, 2019-01-14:
Since this issue is 3 years old now and the code base has moved on, could you re-test on the current master @lukedarnell ? (Or somebody else with access to IUS.)

mciepluc, 2019-01-14:
Test passes for me just now:
```xcelium> run
    0.00ns INFO     cocotb.gpi                                  gpi_embed.c:278  in embed_sim_init                  Running on ncsim(64) version 18.03-s012
    0.00ns INFO     cocotb.gpi                                  gpi_embed.c:279  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
    0.00ns INFO     cocotb                                      __init__.py:132  in _initialise_testbench           Running tests with Cocotb v1.0.1 from /home/mcieplucha/dsp_top/tb/cocotb
    0.00ns INFO     cocotb                                      __init__.py:148  in _initialise_testbench           Seeding Python random module with 1547474535
    0.00ns INFO     cocotb.regression                         regression.py:183  in initialise                      Found test test_mixed_language.mixed_language_test
    0.00ns INFO     cocotb.regression                         regression.py:315  in execute                         Running test 1/1: mixed_language_test
    0.00ns INFO     ..ixed_language_test.0x7f9b7fcb2cf8       decorators.py:209  in send                            Starting test: "mixed_language_test"
                                                                                                                    Description: 
                                                                                                                        Try accessing handles and setting values in a mixed language environment
                                                                                                                        
  100.00ns INFO     cocotb.endian_swapper_mixed        .._mixed_language.py:14   in mixed_language_test             Got: 'i_swapper_sv'
  100.00ns INFO     cocotb.endian_swapper_mixed        .._mixed_language.py:17   in mixed_language_test             Got: 'i_swapper_vhdl'
  300.00ns INFO     cocotb.endian_swapper_mixed        .._mixed_language.py:26   in mixed_language_test             Both signals read as 1
  301.00ns INFO     cocotb.regression                         regression.py:260  in handle_result                   Test Passed: mixed_language_test
  301.00ns INFO     cocotb.regression                         regression.py:206  in tear_down                       Passed 1 tests (0 skipped)
  301.00ns INFO     cocotb.regression                         regression.py:369  in _log_test_summary               *************************************************************************************************
                                                                                                                    ** TEST                                     PASS/FAIL  SIM TIME(NS)  REAL TIME(S)  RATIO(NS/S) **
                                                                                                                    *************************************************************************************************
                                                                                                                    ** test_mixed_language.mixed_language_test    PASS          301.00          0.02     16948.39  **
                                                                                                                    *************************************************************************************************
                                                                                                                    
  301.00ns INFO     cocotb.regression                         regression.py:386  in _log_sim_summary                *************************************************************************************
                                                                                                                    **                                 ERRORS : 0                                      **
                                                                                                                    *************************************************************************************
                                                                                                                    **                               SIM TIME : 301.00 NS                              **
                                                                                                                    **                              REAL TIME : 0.03 S                                 **
                                                                                                                    **                        SIM / REAL TIME : 9842.56 NS/S                           **
                                                                                                                    *************************************************************************************
```

imphil, 2019-01-14:
Fantastic, closing this as fixed. @lukedarnell please reopen if you still experience this issue on your side.

closed 2019-01-14 by imphil