cocotb/cocotb #251: Update QUESTA makefile to support VHDL and FLI interface

leftink, 2015-07-16:
Makefile.questa currently only supports building Verilog.  It needs to be updated to support VHDL and the FLI interface.

I was able to accomplish this by replacing the contents of Makefile.questa with the contents of Makefile.modelsim and then removing the statement "ARCH:=i686" at the top.

Questa supports all the same commands as Modelsim.

leftink, 2015-07-16:
Chris, I looked at the changes, and they look close, but I noticed one issue in the Makefile.questa.

It looks like lines 30 and 31 shouldn't be in the Makefile.questa because QUESTA is not limited to 32-bit.

chiggs, 2015-07-16:
@leftink of course, thanks for pointing that out. Should be fixed in now 832a4ae8c3e1be2ba4faceb5897bb30d2cae6867

commit 3f53f963 (2015-07-17): Merge pull request #254 from potentialventures/issue251

Fixes #251: Harmonise Makefiles between Modelsim and Questa
commit 9c71dee5 (2015-07-17): Fixes #251: Harmonise Makefiles between Modelsim and Questa

* Only difference is that Modelsim is 32-bit only
commit 082e1316 (2015-07-24): Fixes #251: Use full path to include Makefile.questa
commit cc0e437f (2015-07-24): Merge pull request #264 from potentialventures/Issue#251

Fixes #251: Use full path to include Makefile.questa
closed 2015-07-17 by chiggs