cocotb/cocotb #253: Not defining TOPLEVEL does not return the first root object

leftink, 2015-07-16:
The documentation states that if the TOPLEVEL is not defined then the first "root" handle will be used as the DUT.  

In the environment that I am using, we have a custom build system and I am not able to dynamically specify the top at this time.  I thought that by not defining TOPLEVEL, I would be able to workaround the limitation of our build system until it can be fixed.

I found that in "gpi_embed.c::embed_sim_init()", if TOPLEVEL is not defined the initialization fails in contradiction to the documentation.

I tried modifying the code to work around this, but I could never get it to work, because even though I could essentially pass "None" into the python code for initialization, I couldn't figure out how to pass NULL into the "get_root_handle(const char *name) function which is what the function is looking for in order to return the first root handle.

I worked around this, by defining TOPLEVL as an empty string, i.e. set TOPLEVEL "".  I then had to modify the FliImpl::get_root_handle() function to look for an empty string instead of NULL and then it worked as described in the documentation.

Either the documentation needs to be updated to reflect this feature is not supported or some work needs to be done to implement this feature.

stuarthodgson, 2015-07-20:
Not closing as need to add test case

FinnG, 2019-06-27:
It looks like this issue is still present for me - I'm using Icarus Verilog on Linux Mint:

```
gfg@rom ~/git/cocotb $ uname -a
Linux rom 4.15.0-45-generic #48~16.04.1-Ubuntu SMP Tue Jan 29 18:03:48 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
gfg@rom ~/git/cocotb $ cat /etc/issue
Linux Mint 18.3 Sylvia \n \l
```

A simple `make` in the top level of the repo failed, because the `make` invocation in `tests/test_cases/issue_253` was failing as follows:

```
gfg@rom ~/git/cocotb/tests/test_cases/issue_253 $ make
Makefile:44: warning: overriding recipe for target 'results.xml'
/home/gfg/git/cocotb/cocotb/share/makefiles/simulators/Makefile.icarus:80: warning: ignoring old recipe for target 'results.xml'
make results.xml
make[1]: Entering directory '/home/gfg/git/cocotb/tests/test_cases/issue_253'
Makefile:44: warning: overriding recipe for target 'results.xml'
/home/gfg/git/cocotb/cocotb/share/makefiles/simulators/Makefile.icarus:80: warning: ignoring old recipe for target 'results.xml'
Skipping
make[1]: Leaving directory '/home/gfg/git/cocotb/tests/test_cases/issue_253'
PYTHONPATH=/home/gfg/git/cocotb/tests/test_cases/issue_253/build/libs/x86_64:/home/gfg/git/cocotb:/home/gfg/git/cocotb/tests/test_cases/issue_253: LD_LIBRARY_PATH=/home/gfg/git/cocotb/tests/test_cases/issue_253/build/libs/x86_64::/usr/lib:/usr/lib MODULE=issue_253 \
COCOTB_SIM=1 TESTCASE=issue_253_empty TOPLEVEL="" \
vvp -M /home/gfg/git/cocotb/tests/test_cases/issue_253/build/libs/x86_64 -m gpivpi sim_build/sim.vvp   
     -.--ns INFO     cocotb.gpi                                  gpi_embed.c:111  in embed_init_python               Did not detect Python virtual environment. Using system-wide Python interpreter.
     -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:276  in embed_sim_init                  Running on Icarus Verilog version 11.0 (devel)
     0.00ns INFO     cocotb.gpi                                  gpi_embed.c:277  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
     0.00ns INFO     cocotb                                      __init__.py:131  in _initialise_testbench           Running tests with Cocotb v1.1.0 from /home/gfg/git/cocotb
     0.00ns INFO     cocotb                                      __init__.py:148  in _initialise_testbench           Seeding Python random module with 1561668890
Traceback (most recent call last):
  File "/home/gfg/git/cocotb/cocotb/__init__.py", line 168, in _initialise_testbench
    regression_manager.initialise()
  File "/home/gfg/git/cocotb/cocotb/regression.py", line 122, in initialise
    self._dut = cocotb.handle.SimHandle(handle) if handle else None
  File "/home/gfg/git/cocotb/cocotb/handle.py", line 778, in SimHandle
    raise TestError("Couldn't find a matching object for GPI type %d (path=%s)" % (t, path))
cocotb.result.TestError: Couldn't find a matching object for GPI type 0 (path=None)
Cocotb initialisation failed - exiting
     0.00ns ERROR    cocotb.scheduler                            __init__.py:191  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
VCD info: dumpfile waveform.vcd opened for output.
mkdir -p empty_top_level_result && mv results.xml empty_top_level_result/
mv: cannot stat 'results.xml': No such file or directory
Makefile:53: recipe for target 'empty_top_level' failed
make: *** [empty_top_level] Error 1
```

I don't know what simulator @leftink was using, but I'd guess from the mentions of FLI that it wasn't Icarus, which uses VPI. Digging a bit deeper it seems that `VpiImpl::get_root_handle()` was returning  a root handle of type `GPI_NONE`, which causes the crash down the line:

```
cocotb.result.TestError: Couldn't find a matching object for GPI type 0 (path=None)
```

"GPI type 0" is `GPI_NONE`. It seems that there are several top level handles available in Icarus, but not all of them are modules. Adding a check to `VpiImpl::get_root_handle()` to ensure it's returning an object handle of type `GPI_MODULE` (otherwise returning nothing) fixed this issue for me.

I will submit a pull request with my change in but my knowledge of VPI is very rusty, so I'm not sure if this is the correct long term solution. Feel free to reject it if there is a more appropriate solution.

imphil, 2019-06-29:
Thanks @FinnG for reporting this!

This is actually a regression/behavior change in iverilog. It still works in 10.1, but doesn't work any more in 11 (devel). To keep this issue focused, I've moved the discussion to #982.

The original issue here is actually resolved, I'm closing this issue to reflect it.

closed 2019-06-29 by imphil

ktbarrett, 2026-01-08:
This behavior broke at some point. Setting `TOPLEVEL=""` no longer gets the root handle, only `nullptr` does.