cocotb/cocotb #255: Methods for driving signals in Python are not consistent

leftink, 2015-07-16:
In testing, I have found there are two consistent ways to drive signals and one way that only works in certain instances.

Consistent ways:
1.  dut.clk.value = 1
2.  dut.clk <= 1 (shorthand for the first way)

Inconsistent way:
1.  obj.sig = 1 (only works if obj is a parent handle, i.e. dut.clk = 1)

The "inconsistent way" only works if referenced from a parent handle.  If a bus is created with signals, the following will not work:
    bus.enable = 1

The 'enable' will not be driven to a '1' using the equal, one must use one of the two consistent ways.

Another example that shows the inconsistent way and how it is confusing is:
clk = dut.clk

dut.clk = 1 ##drives the clock signal to one
clk <= 0 ##drives the clock signal to zero
clk = 1 ##does not drive the clock signal.  If clk truly is dut.clk then any operation I do on dut.clk should be able to be done to the variable clk.

I looked into why the inconsistent way works and it is because handle.py supports the __setattr__ function.  I would recommend not overriding this function.  There isn't a need for it that I can see.  Then the user will have two consistent ways of driving a signal.

chiggs, 2015-07-16:
@leftnik  There was some discussion of this on #119 and also #134.

Unfortunately there isn't really a nice way to do this consistently.  Simple assignment with `__setattr__` is neatest, but only works for hierarchical reference.  In Python there's no assignment operator overload so `clk = 1` will always re-bind the variable name `clk` to an integer.

Overloading `<=` is at least consistently applicable, but has drawn complaints from some areas for being un-Pythonic.  Other options would be to provide a `__call__` method so that `clk(1)` did the assignment.

It would obviously be better to have a single consistent way to perform assignments, however it seems we haven't yet reached agreement on what that is.

leftink, 2016-04-01:
@stuarthodgson I agree with everything said here.  I think this can be closed as a "no fix", but I leave it up to you guys.

closed 2016-09-14 by leftink