cocotb/cocotb #256: Hierarchical Naming does not work with the FLI Interface

leftink, 2015-07-16:
I tried to access a signal in a lower component and the FLI interface failed to find it, E.g.  dut.comp_a.sig_b.value.

The FLI cannot find comp_a in the design.

Comparing the implementation to the VHPI implementation, I think there needs to be more in the FLI implementation.

Looking in the "native_check_create()" function, VHPI gets the handle by calling "vhpi_handle_by_name()" which in my reading seems to get a handle to any type of element by the name provided.

Looking in the "native_check_create()" function, FLI gets the handle by calling "mti_FindSignal()" which in my reading will ONLY return a handle to a signal and not to any element in the hierarchy.

I believe to get the same functionality as in VHPI, the function will have to use three functions and do trial and error, i.e.:
  1.  call mti_FindRegion() to check if the name is a region, return handle if not NULL
  2.  call mti_FindSignal() to check if the name is a signal, return handle if not NULL
  3.  call mti_FindVar() to check if the name is a variable, generic, or constant, return handle if not NULL
  4.  return NULL if all three fail to find the name.

stuarthodgson, 2015-07-20:
Not closing for now as need to add some regressions to test for these

leftink, 2016-04-01:
@stuarthodgson test_array fully exercises this.  I think this can be closed if you agree.

closed 2016-04-01 by stuarthodgson