cocotb/cocotb #257: LOG_DEBUG call in FLI causes bad handle in Questa

leftink, 2015-07-17:
Using QuestaSIM and enabling DEBUG logging for the "cocotb.gpi" logger causes simulations to crash with a bad handle exception.

I tracked the issue to "FliImpl.cpp::FliSignalObjHdl::get_signal_value_binstr()".

The error is in the last LOG_DEBUG call.

Current:
   LOG_DEBUG("Retrieved \"%s\" for signal %s", &m_val_buff, m_name.c_str());

Fix:
   LOG_DEBUG("Retrieved \"%s\" for signal %s", m_val_buff, m_name.c_str());

Should just pass m_val_buff because it is already a pointer.

stuarthodgson, 2015-07-20:
Not closing this for now as we need to add a regression that runs tests with LOG_DEBUG level

ktbarrett, 2020-05-05:
The method does not exist any more and all implementations of `get_signal_value_binstr` in FLI look correct.

closed 2020-05-05 by ktbarrett