cocotb/cocotb #258: Split regression tests out from examples

chiggs, 2015-07-17:
Cocotb currently has a mixture of examples and functionality tests all under `examples/*`

Propose changing the directory structure to introduce `tests` at the toplevel for all regression tests and reduce examples to a small number of (well documented) examples.

stuarthodgson, 2015-07-17:
Agreed. Www should also where possible have a test_xxxx where xxxx is a bug number. Not all bugs will have tests but those tar do should relate back.

saturn77, 2015-07-17:
I have been working on a whitepaper that explains how to use coco_tb with a simple digital IIR filter that I could share.

chiggs, 2015-07-18:
@saturn77 that would be great - documented examples are always helpful!

closed 2015-07-29 by stuarthodgson