cocotb/cocotb #267: Driving a Binary String through the FLI is dependent on simulator settings

leftink, 2015-07-22:
In the function, FliImpl.cpp::set_sign_value(std::string &value), the call to mti_ForceSignal() passes a string representing the values for every position in the vector.

Apparently, in Mentor's infinite wisdom, the interpretation of that string through this interface is dependent on the "radix" that is set for that signal, or if not set, the global (default) radix.

In my case I was trying to drive a 32-bit std_logic_vector, and by default our "radix" is set to hex, and to my surprise the value driven into the signal was the last 8-bits of the string treated as a 32-bit hex string.

The only solution to this problem, that would let me drive a std_logic_vector with values other than 0 and 1, was to modify the string format.  The following are some examples of the string format:
    string = "8'b01101000"
    string = "10'bZ1XU000011"

I tested the solution using the FLI interface in Questa.

chiggs, 2015-07-22:
Wow, that's an interesting API "feature".

Is there a way to retrieve the radix that is applied to a signal?  This can be changed at run time in the GUI, so I assume we'd have to re-check every time...

leftink, 2015-07-22:
I didn't see anyway to get the radix, but if you use the new string format, it drives it properly regardless of the radix.

commit b7468a16 (2015-08-05): Fixes #267: Removes dependency on simulator radix setting when sending binary string
commit d45a67eb (2015-08-05): Merge pull request #277 from leftink/issue267

Fixes #267: Removes dependency on simulator radix setting when...
closed 2015-08-05 by stuarthodgson