cocotb/cocotb #268: 32-bit signals are driven by binary string and not the integer value

leftink, 2015-07-22:
In handle.py, the function setimmediatevalue() will check if the value is an integer and then check if the integer is less than 0x7fffffff and then it will drive the simulator quicker if it is.  It seems like any 32-bit value should be able to be driven in the same manner.  I realize that that if the value uses the most significant bit, then the C-function, will treat that value as a negative integer, but from the code I have looked at, there wouldn't be any impact to the way the signal is driven.

@chiggs Is there a reason that any 32-bit and below values don't use the faster interface, instead of 31-bit and below?  It seems like the C-Interface would support this.

chiggs, 2015-07-22:
It likely stems from the VPI definition of a value, which by the specification is an `int32_t`.

It would be difficult to prove that all simulators treated this as `uint32_t`, however VHPI uses `uint32_t` and FLI arguably doesn't matter - we should even default to `uint64_t` before switching to strings.

It would be nice if in Python, we determined the threshold for switching to string-based slow path when the signal object is discovered, based on some constant we could read from the interface.

stuarthodgson, 2015-07-22:
We can do that with a capabilities method that python can call. Would be extensible and www can have some sensible defaults that are inplementation dependent.

imphil, 2019-11-20:
The VPI definition of int32_t comes most likely from the fact that int/integer in SystemVerilog are defined as signed 32 bit type. So the `IntegerObject` should at least support the negative range as well.

However, as far as I could see, the `unsigned` qualifier is not reflected in VPI; both `logic int unsigned something` and `logic int something` are most likely mapped to `vpiIntegerNet`. This means we can safely cast uint32 to int32 and set it through `simulator.set_signal_val_long()`.

Needs more research, but could be a very nice performance win for cocotb.

elgorwi, 2021-01-09:
Fixed by #2316

PR #2316 (2021-01-09): set_signal_value_long() bugfixes and improvements
Some bugfixes and improvements for setting signals using python ints: 
closes #1202 
closes #913 
closes #268 

This PR is essentially a merge of PR #1792 , PR #1742  and #1744 .
For more details refer to the linked issues and PRs.
closed 2021-01-09 by ktbarrett