cocotb/cocotb #274: Multiple top-level modules with same name and different library

amal-khailtash, 2015-07-28:
It would be great to use library-name/module-name to refer to the module to use for testing instead of dut.  There could be multiple top-level modules with the same name in different libraries that need to be simulated for the purpose of functional equivalency check.

amal-khailtash, 2015-07-28:
A way to access modules within a library other than work.  If there are multiple top-levels on vsim command line, signals cannot be found even in work library.

    vsim work.module1 another_lib.module2
    work.module1 contains no object named clock

How do I access objects under another_lib.module2?

ktbarrett, 2020-05-05:
In this case, simply create a top-level object that instances both of the modules you are interested in.

closed 2020-05-05 by ktbarrett