cocotb/cocotb #280: Enhance the Test Class to support pre- and post- test calls

leftink, 2015-07-29:
So I was trying to be fancy with one of my tests, and support setting a particular seed for a particular test.  In the same module I have a test factory that has several other tests.  In the test, I save the current random state and then seed it with the desired value and then at the end of the test, restore the original state.  This allows the factory tests to be different from run to run even if a test is run that wants to use a certain seed.

My approach worked great as long as the test ran without any failures, but if a component stops the test early, e.g. the Scoreboard, my code to restore the random state doesn't get executed which means all subsequent tests are always the same from run to run.

I was thinking it might be nice if you could assign a pre- and post- functions to the test, that are always run with pre- running before each test and post- running after each test even if it fails early.

chiggs, 2015-07-29:
One of my longer-term aspirations is to enable tests to be written using the `unittest` module from the Python standard library, which would support `setUp` and `tearDown` both per-test and per test fixture.

leftink, 2016-05-24:
@chiggs Have you seen VUNIT [https://github.com/VUnit/vunit](url)?  This might be an interesting integration.  It even has file dependency processing.

jarickc, 2016-07-13:
Vunit is fairly well done and would probably be the best starting point for any unit test for VHDL/SV TB work. I would love to see these two projects integrated a bit better personally.

ktbarrett, 2021-01-30:
I split off the talk about VUnit.

Personally I don't like the idea of having to add more functionality to the regression manager since we may replace it with pytest in the future. However, this is currently *necessary* due to erroring forked coroutines killing the test without going through normal code channels; meaning a context manager at the top of the test won't work.

ktbarrett, 2021-08-20:
`cocotb.test()` and `TestFactory` now support wrapping any callable that returns a coroutine as the test function. It should be trivial to add pre and post test hooks now.

closed 2021-08-20 by ktbarrett

joaopccarvalho, 2023-12-05:
> `cocotb.test()` and `TestFactory` now support wrapping any callable that returns a coroutine as the test function. It should be trivial to add pre and post test hooks now.

How to run a post-action after the factory.generate_tests()? For example, how to export coverage only after all the Factory Tests and not for each one?