cocotb/cocotb #281: Enhance the boot sequence to support a Configuration File

leftink, 2015-07-29:
Where I work, we use a proprietary build system instead of "make" and because of this it is very difficult to use environment variables that are changing from run to run.  It likes a static environment because the tool always runs in a clean environment with only pre-defined environment settings.

I was thinking, we could update the boot sequence to support reading in a "configuration" file from the run directory and setting up the environment accordingly.  If the file is not present, it could still fall back to seeing if the appropriate environment variable is set.

stuarthodgson, 2015-07-29:
@leftink . Quite like this, Woudl also give a change to document all the environment vars that are used and what they do. Maybe also be able to have required/optional ones as well and do nice errors up front rather than at a point in the future.

However this would be pretty low down the list of features to do for the moment.

chiggs, 2015-07-29:
Included in the config file should be the Python module (or list of modules) to run - see #273 which I have closed.

ktbarrett, 2020-05-05:
This is related to #1225.

ktbarrett, 2025-03-27:
cocotb now has `PYGPI_USERS` which can be used to load in any arbitrary Python from the GPI Python wrapper. You can set this to load into a configuration file before loading into the RegressionManager. The startup and shutdown routines in `cocotb._init` could be cleaned up and segregated, but that will wait until there's some actual user feedback on what would be beneficial.

closed 2025-03-27 by ktbarrett