cocotb/cocotb #282: [FLI] Older Modelsim errors when driving signals

schribl, 2015-07-30:
We use an older Modelsim version (10.0a) and encountered two issues.

### Driving enumerations

When trying to set an integer value to an enumeration signal (like std_logic) cocotb calls mti_ForceSignal with a value string like "16#00000001". However, the mentioned Modelsim version does not like that:

    # ** Error: (vsim-4026) Value "16#00000001" does not represent a literal of the enumeration type.
    # ** Error: (vsim-4011) Invalid force value: 16#00000001.

This error is encountered in e.g. clock generation. I solved this issue with a quick and dirty hack in FliImpl.cpp:

```c
int FliSignalObjHdl::set_signal_value(const int value)
{
    int rc;
    char buff[20];
    mtiTypeKindT kind;

    kind = mti_GetTypeKind(mti_GetSignalType(m_fli_hdl));
    if (kind == MTI_TYPE_ENUM) {
        snprintf(buff, 20, "%1i", value);
    } else {
        snprintf(buff, 20, "16#%016X", value);
    }

    rc = mti_ForceSignal(m_fli_hdl, &buff[0], 0, MTI_FORCE_DEPOSIT, -1, -1);

    if (!rc) {
        LOG_CRITICAL("Setting signal value failed!\n");
    }
    return rc-1;
}
```

When the signal is an enumeration its value is set as string instead of a integer with radix.

### String length

While testing cocotb I also encountered also that for example avalon.py in _send_string creates a BinaryValue with width X and use it to drive several signals with different widths. Then Modelsim throws an error stating the width do not match. In my opinion this is not an issue in the FLI interface but the user code should only use the correct width. However, I thought it might be worth noting.

leftink, 2016-06-28:
@schribl and @stuarthodgson I don't know if this has been tried recently, but I would venture to guess this can be closed as the interface mentioned has been completely re-written and mti_ForceSignal() is no longer used.

schribl, 2016-06-29:
I updated my local cocotb installation to the current master branch. After some minor tweaks to my tests everything worked out of the box as expected. So @leftink is right and this issue can be closed.

closed 2016-06-29 by schribl