cocotb/cocotb #291: EXTRA_ARGS not passed to Questa vsim

mciepluc, 2015-08-22:
Hello,

EXTRA_ARGS var is not passed to the vsim args when using Questa Simulator.

LinuxFanatic777, 2016-03-11:
I just started to use cocotb and have same problem: EXTRA_ARGS is not passed to simulator.
Inside Makefile.questa I clearly see that EXTRA_ARGS is not passed to vsim.

This option highly required for designs that uses libraries from FPGA-manufacturer to simulate, so I have to add this manually into Makefile.questa file...

mciepluc, 2016-03-11:
I am quite sure I fixed that.
Please check if you are up to date with recent master branch. The code in Makefile.questa:
```
ifdef EXTRA_ARGS
VLOG_ARGS += $(EXTRA_ARGS)
endif
```

LinuxFanatic777, 2016-03-11:
Excuse me, I'm dumb noob =))) EXTRA_ARGS is for Verilog compiler (which is correct with cocotb), while VSIM_ARGS is what I actually want. That's how I solved problem:

`export VSIM_ARGS=" -L altera_mf_ver -L lpm_ver "
linux32 make ARCH=i686 SIM=modelsim`

VSIM_ARGS works perfectly!

potentialventures, 2016-03-11:
It does highlight a slight discrepancy in the docs / Makefiles which needs fixing.

We should have `COMPILE_ARGS` passed to the compilation stage, `SIM_ARGS` passed to the simulation stage and `EXTRA_ARGS` passed to both.  This avoids simulator specific commands such as `VSIM_ARGS` which makes it harder to create environments that will be portable between simulators.

Unfortunately right now the makefiles don't all conform to this pattern so this is a bug that does need addressing.

mciepluc, 2018-02-19:
The problem is that simulators handle compilation/elaboration and simulation phases different way. 
For Icarus and VCS it is as above, ```COMPILE_ARGS``` and ```EXTRA_ARGS``` appended for compilation command and ```SIM_ARGS``` and ```EXTRA_ARGS``` for simulation.
Cadence IUS is one-step, so it can be done ```EXTRA_ARGS += COMPILE_ARGS + SIM ARGS``` (Currently only ```EXTRA_ARGS``` works). 

But for Questa it is a bit more difficult, as there are 3 phases, including also a call for the Questa application run. I suggest introducing a ```BIN_EXTRA_ARGS``` to fix this for Questa, and maybe it will be useful in the future for other simulators.  So ```BIN_EXTRA_ARGS``` will be passed the same way as ```EXTRA_ARGS``` for other simulators, except Questa, where it will be passed only to the vsim command this way:
 ```vsim $(BIN_EXTRA_ARGS) -do runsim.do```

It seems Aldec works the same way, but I have no option to test it. Could any Aldec user comment on that?

mciepluc, 2018-11-15:
I am going to clean this issue to provide a uniform Makefiles syntax. 
Given that simulators may have different syntax for compilation/simulation args we can either (1) differentiate simulator-specific compilation/simulation args (e.g. by introducing ```VSIM_ARGS```, ```VCS_ARGS```, ```IRUN_ARGS```...) or provide a uniform syntax (2). An example testbench makefile would be (assume we use only Icarus and Questa):
```
(1)
...
ICOM_ARGS += -DFOO
VCOM_ARGS  += +define+FOO
...
(2)
...
ifeq($(SIM),questa)
COMPILE_ARGS += +define+FOO
else
COMPILE_ARGS += -DFOO
endif
...
```

The same must be done for ```BIN_EXTRA_ARGS``` as explained in my previous comment.
Please provide some feedback which version you thinks is better.

themperek, 2018-11-15:
>Please provide some feedback which version you thinks is better.

Similar to me sightly going into direction of (2)

closed 2018-11-28 by mciepluc