cocotb/cocotb #293: Ability to rename bus signals for a busdriver

Martoni, 2015-08-23:
In the endian_swapper example, avalonMaster bus is used like it : 
```python
 self.csr = AvalonMaster(dut, "csr", dut.clk)
```

That work only if avalon signals has exactly these names in HDL top component : 
```vhdl
        csr_address             : in    std_ulogic_vector(1 downto 0);
        csr_readdata            : out   std_ulogic_vector(31 downto 0);
        csr_readdatavalid       : out   std_ulogic;
        csr_read                : in    std_ulogic;
        csr_write               : in    std_ulogic;
        csr_waitrequest         : out   std_ulogic;
        csr_writedata           : in    std_ulogic_vector(31 downto 0)
```

Is there a solution to rename signals when AvalonMaster object is instantiated ? If we can't change the HDL signals name it can be useful.

chiggs, 2015-08-23:
There isn't an "official" way, perhaps this should be added to BusDriver.

As a short-term fix it's possible to hack around this by adding a signal alias:

```
# Add alias mapping csr_address to csr_addr
dut._sub_handles["csr_address"] = dut.csr_addr

self.csr = AvalonMaster(dut, "csr", dut.clk)
```

Martoni, 2015-08-24:
Great ! That exactly I was looking for. Thanks.

closed 2015-08-24 by Martoni