cocotb/cocotb #304: endian_swapper with ghdl looping forever with no output.

svenn71, 2015-08-29:
Hi,
I have just downloaded cocotb to evaluate it as a better way to test my vhdl code. Starting with the endian_swapper as it has vhdl code. Using ghdl 0.31 64-bit on Debian I started the simulation with make SIM=ghdl GPI_IMPL=vhpi and cocotb fired up the compilation process which ended in a process running without any output. Processor load show me that something is running, but nothing is output to the screen. I cleaned up and tested with iverilog as a reference. This run yielded monitor output and a waveform file as expected.

How mature is the integration of ghdl in cocotb? Something still missing? Most of my designs are in VHDL and I would love to find an alternative way to writing all testbenches in plain VHDL.

chiggs, 2015-08-29:
Hi @svenn71,

Sadly at this point in time, I don't believe there is a free VHDL simulator that supports enough of the VHPI standard required for Cocotb.  We did briefly look into [Cocotb on nvc](https://github.com/nickg/nvc/issues/64) but this needs further work.  There has been some activity on GHDL recent but I doubt there is any appetite for expanding the VHPI implementation.

If you have access to commercial simulators, either Synopsys VCS or Aldec Riviera-PRO work with VHDL designs.

Thanks,

Chris

svenn71, 2015-08-30:
Ok, fair enough. Good to know that it is on the ghdl side work has to be done. I'll see what can be done on that side.

chiggs, 2015-09-11:
@svenn71 I've changed the title to more accurately reflect the issue and marked as wontfix, meaning the work to do is on the GHDL side rather than in Cocotb.

Dolu1990, 2016-06-06:
Hi,
Any update about this thread ?
It also look like the ghdl makefile in cocotb is broken

cd $(SIM_BUILD) && $(CMD) -a $(VHDL_SOURCES) && $(CMD) -e $(TOPLEVEL)
should be
cd $(SIM_BUILD) && $(CMD) -a $(VHDL_SOURCES) && $(GHDL) -e $(TOPLEVEL)

right ?

JarrettR, 2016-07-12:
GHDL works pretty well with cocotb now thanks to excellent work by @tgingold, close issue?

svenn71, 2016-07-12:
close issue.

Fatsie, 2017-05-17:
With which version of ghdl is cocotb supposed to run ? I am using recent git version of cocotb and ghdl 0.33. I get following error for my testbench:

    ...
    cd sim_build; \
    PYTHONPATH=/home/verhaegs/eda/code/cocotb/build/libs/x86_64:/home/verhaegs/eda/code/cocotb:/home/verhaegs/eda/code/c4m_jtag/sim/cocotb: LD_LIBRARY_PATH=/home/verhaegs/eda/code/cocotb/build/libs/x86_64:/home/verhaegs/software/lib:/home/verhaegs/software/lib64:/home/verhaegs/software/lib:/home/verhaegs/software/lib64:/home/verhaegs/software/lib:/home/verhaegs/software/lib64::/usr/lib64:/usr/lib64:/usr/lib64:/usr/lib64 MODULE=test_idcode \
        TESTCASE= TOPLEVEL=c4m_jtag_tap_controller TOPLEVEL_LANG=vhdl \
    /home/verhaegs/software/bin/ghdl -r  c4m_jtag_tap_controller --vpi=/home/verhaegs/eda/code/cocotb/build/libs/x86_64/libvpi.so 
    loading VPI module '/home/verhaegs/eda/code/cocotb/build/libs/x86_64/libvpi.so'
         -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:91   in gpi_print_registered_impl       VPI registered
    vpi_register_cb: unknwon reason
     -.--ns ERROR    cocotb.gpi                                 VpiCbHdl.cpp:74   in arm_callback                    VPI: Unable to register a callback handle for VPI type cbStartOfSimulation(11)
    make[1]: *** [results.xml] Fout 255

svenn71, 2017-05-18:
I compiled ghdl from Tristans github repository.

Martoni, 2017-05-18:
I'm using this version (from github repository):

> GHDL 0.34-dev (2017-03-01-58-g3031cd3) [Dunoon edition]

And that works.

Fatsie, 2017-05-20:
Compiled now latest git version and it seems to work.

closed 2019-01-24 by imphil