cocotb/cocotb #308: Problem with SIM=Questa (FliImpl.h:32:17: fatal error: mti.h: No such file or directory)

OrionBytes, 2015-09-04:
./examples/adder/tests/
make SIM=Questa 
results in 
>FliImpl.h:32:17: fatal error: mti.h: No such file or directory
> #include "mti.h"
> compilation terminated.

When just make - all is ok.
What can be wrong?
About a 2 weeks ago all seems to be ok with Cocotb + Questasim.
Now it's a new system (OpenSuse 64) but Questasim works good - compiles adder.v successfully.

stuarthodgson, 2015-09-04:
@OrionBytes , this error indicates that cocotb is not able to find the mti.h header file which is located in your Questa installation directory. We find that by picking up the location of `vsim` from the $PATH environment variable. 

So things to check.

1. Is the location of `vsim` on your PATH and where you expect it to be?
    
    `which vsim`
2. Is there an mti.h inside your installation of Questa, 
   
     `ls -l $(dirname $(which vsim))/../include/ | grep mti`

The answer to both of these must be yet, If not then you need to fix your environment or installation of Questa.

chiggs, 2015-09-04:
@stuarthodgson I think the FLI compilation should be optional, it's not necessarily guaranteed to be present.

We also may have broken Modelsim by assuming the FLI headers are available:  https://travis-ci.org/chiggs/travis_experiment/builds/77464232

stuarthodgson, 2015-09-04:
@chiggs 9c71dee introduced this change to harmonize Questa and Modelsim makefiles. I'm sure (but can not find it) we also had a change where FLI was enabled for modelsim and thus Questa as the headers where always present even if the license was not. On the versions of modelsim I have installed, including the free version I have the headers. Maybe something has changed.

OrionBytes, 2015-09-04:
Yes, the problem was around correct PATH.
Now QuestaSim started but with new error 
(seems similar to https://github.com/potentialventures/cocotb/issues/217):
# 10.2c
# do runsim.do 
# QuestaSim-64 vlog 10.2c Compiler 2013.07 Jul 18 2013
# -- Compiling module adder
# 
# Top level modules:
# 	adder
# vsim -onfinish exit -pli libvpi.so adder 
# ** Note: (vsim-3812) Design is being optimized...
# ** Note: (vsim-3865) Due to PLI being present, full design access is being specified.
# Loading libvpi.so
# ** Error: (vsim-3197) Load of "libvpi.so" failed: libvpi.so: cannot open shared object file: No such file or directory.
# ** Error: (vsim-PLI-3002) Failed to load PLI object file "libvpi.so".

Tomorrow I'll try to understand what is going on.
With the previous version of the QuestaSim (and older OpenSuse) all was ok before.

OrionBytes, 2015-09-20:
I forgot to write a quick dirty solution for this problem  (and #217, as I understand):- 
If questasim falls into 
** Error: (vsim-PLI-3002) Failed to load PLI object file "libvpi.so".
with original cocotb makefiles
but all .so libraries exist in /cocotb/build/libs/_your arch_/
it's possible to correct through 
ln -s /cocotb/build/libs/_your arch_/  /_system libraries path_/
! but it's very bad solution !

seems to be some problems in makefiles for questasim
Opensuse 64, QuestaSim-64 10.4a Compiler 2015.03

full log of "make SIM=questa" after "make clean" (adder example) - 
http://pastebin.com/Nq7DxDPT

kgugala, 2015-10-16:
you can also workaround the lib issue by setting LD_LIBRARY_PATH before calling make e.g:

LD_LIBRARY_PATH=/home/karol/antmicro/cocotb/build/libs/x86_64 SIM=questa make

cmarqu, 2019-08-19:
Re the ``mti.h`` problem - apparently there are some installations where ``vsim`` is a wrapper or a symlink so the logic used for determining the ``include/`` directory fails.

An alternative approach would be:
- create a fresh temporary directory so that it is certain that no user's ``modelsim.ini`` is present
- from this directory, call ``vmap std`` which will use the installation's ``modelsim.ini`` to print the location of the VHDL std library
- the base directory of this library path can be used to append ``/include`` to find ``mti.h``

``cocotb-test`` (see #1062) does not require ``mti.h`` however, so maybe it's not worth implementing this approach.

themperek, 2019-08-20:
`mti.h` looks to be needed for `FLI` on need link with `mtipli` so the path to installation is needed. 

In special cases, you cannot specify `MODELSIM_BIN_DIR`?

cmarqu, 2019-08-20:
Making things work automatically is always nicer :)

themperek, 2020-04-16:
There ware changes to how this works (currently in development version) please also see: 
https://docs.cocotb.org/en/latest/simulator_support.html#sim-modelsim

This issue is from 2015 so I will close. Please reopen if the issue is still there.

closed 2020-04-16 by themperek