cocotb/cocotb #310: If module set in MODULE variable not found, error message is not clear

chevillotn, 2015-09-07:
For example if MODULE is set to 'fpga_tb1' and there is no fpga_tb1.py file, the output is not really clear, see below. Actually the call from gpi_embed.c:
cocotb_retval = PyObject_CallObject(cocotb_init, cocotb_args);

does not return. Is there a possibility that this call returns if the python script causes an exception?
This would allow to go to the path that outputs an error message:
fprintf(stderr,"Cocotb initialisation failed - exiting\n");

LD_LIBRARY_PATH=/userselec/chevillot/Projects/ATLAS/git/cocotb/atlas-lar-ldpb-firmware/LATOME/../env/cocotb/build/libs/i686/python2.7:/usr/lib:/usr/lib:/usr/lib:/usr/lib \
        MODULE=fpga_tb1 \
        TESTCASE= \
        TOPLEVEL=fpga_tb_v \
        PYTHONPATH=/userselec/chevillot/Projects/ATLAS/git/cocotb/atlas-lar-ldpb-firmware/LATOME/../env/cocotb/build/libs/i686/python2.7:/userselec/chevillot/Projects/ATLAS/git/cocotb/atlas-lar-ldpb-firmware/LATOME/../env/cocotb:/userselec/chevillot/Projects/ATLAS/git/cocotb/atlas-lar-ldpb-firmware/LATOME/src/fpga/test/testbench \
        /data/questasim/linux_x86_64/vsim -pli libvpi.so -L altera_ver -L lpm_ver -L sgate_ver -L altera_mf_ver -L altera_lnsim_ver -L twentynm_ver -L twentynm_hssi_ver -L twentynm_hip_ver -L altera_common_sv_packages -L tse_mac_phy_altera_reset_controller_150 -L tse_mac_phy_altera_eth_tse_nf_phyip_terminator_150 -L tse_mac_phy_altera_xcvr_native_a10_150 -L tse_mac_phy_altera_eth_tse_pcs_pma_nf_phyip_150 -L xcvr_rx_phy_12ch_altera_xcvr_native_a10_150 -L altera -L lpm -L sgate -L altera_mf -L altera_lnsim -L twentynm -L twentynm_hssi -L twentynm_hip -L xcvr_tx_atxpll_12ch_altera_xcvr_atx_pll_a10_150 -L tse_mac_phy_altera_eth_tse_avalon_arbiter_150 -L tse_mac_phy_altera_eth_tse_mac_150 -L xcvr_tx_reset_12ch_altera_xcvr_reset_control_150 -L tse_reset_altera_xcvr_reset_control_150 -L xcvr_tx_phy_12ch_altera_xcvr_native_a10_150 -L iopll_160_altera_iopll_150 -L lli_altera -L clk_ctrl_altclkctrl_150 -L istage_altera -L tse_atxpll_altera_xcvr_atx_pll_a10_150 -L xcvr_rx_reset_12ch_altera_xcvr_reset_control_150 -L fifo_fifo_150 -L Pattern_ram_ram_2port_150 -L fpga_tb -L tse_mac_phy_altera_eth_tse_150 -L fpga -L istage -L bitvis_util -L ipctrl -L lli -L user -L remap -L ttc -L osum -L mon -L testbench -L work -l logs/vsim.log -wlf logs/vsim.wlf -error 3473 -voptargs=+acc -multisource_delay latest -t ps +typdelays -c -do logs/vsim_sim.do "fpga_tb.fpga_tb_v"
Reading pref.tcl

# 10.2c_5

# vsim -L altera_ver -L lpm_ver -L sgate_ver -L altera_mf_ver -L altera_lnsim_ver -L twentynm_ver -L twentynm_hssi_ver -L twentynm_hip_ver -L altera_common_sv_packages -L tse_mac_phy_altera_reset_controller_150 -L tse_mac_phy_altera_eth_tse_nf_phyip_terminator_150 -L tse_mac_phy_altera_xcvr_native_a10_150 -L tse_mac_phy_altera_eth_tse_pcs_pma_nf_phyip_150 -L xcvr_rx_phy_12ch_altera_xcvr_native_a10_150 -L altera -L lpm -L sgate -L altera_mf -L altera_lnsim -L twentynm -L twentynm_hssi -L twentynm_hip -L xcvr_tx_atxpll_12ch_altera_xcvr_atx_pll_a10_150 -L tse_mac_phy_altera_eth_tse_avalon_arbiter_150 -L tse_mac_phy_altera_eth_tse_mac_150 -L xcvr_tx_reset_12ch_altera_xcvr_reset_control_150 -L tse_reset_altera_xcvr_reset_control_150 -L xcvr_tx_phy_12ch_altera_xcvr_native_a10_150 -L iopll_160_altera_iopll_150 -L lli_altera -L clk_ctrl_altclkctrl_150 -L istage_altera -L tse_atxpll_altera_xcvr_atx_pll_a10_150 -L xcvr_rx_reset_12ch_altera_xcvr_reset_control_150 -L fifo_fifo_150 -L Pattern_ram_ram_2port_150 -L fpga_tb -L tse_mac_phy_altera_eth_tse_150 -L fpga -L istage -L bitvis_util -L ipctrl -L lli -L user -L remap -L ttc -L osum -L mon -L testbench -L work -do logs/vsim_sim.do -error 3473 -l logs/vsim.log -voptargs=+acc -c -multisource_delay latest -pli libvpi.so -t ps -wlf logs/vsim.wlf +typdelays fpga_tb.fpga_tb_v
# ** Note: (vsim-3812) Design is being optimized...
# ** Note: (vsim-3865) Due to PLI being present, full design access is being specified.
# Loading /userselec/chevillot/Projects/ATLAS/git/cocotb/atlas-lar-ldpb-firmware/LATOME/../env/cocotb/build/libs/i686/python2.7/libvpi.so
# //  Questa Sim-64
# //  Version 10.2c_5 linux_x86_64 Nov 19 2013
# //
# //  Copyright 1991-2013 Mentor Graphics Corporation
# //  All Rights Reserved.
# //
# //  THIS WORK CONTAINS TRADE SECRET AND PROPRIETARY INFORMATION
# //  WHICH IS THE PROPERTY OF MENTOR GRAPHICS CORPORATION OR ITS
# //  LICENSORS AND IS SUBJECT TO LICENSE TERMS.
# //
#      -.--ns INFO     cocotb.gpi                                GpiCommon.cpp:47   in gpi_print_registered_impl       VPI registered
# Loading work.fpga_tb_v(fast)
# do logs/vsim_sim.do
# ** Error: (vish-4014) No objects found matching '/fpga_tb/fpga/reset_n'.
# Executing ONERROR command at macro ./logs/vsim_sim.do line 1
# ** Error: (vish-4014) No objects found matching '/*'.
# Executing ONERROR command at macro ./logs/vsim_sim.do line 2
Traceback (most recent call last):
  File "/userselec/chevillot/Projects/ATLAS/git/cocotb/atlas-lar-ldpb-firmware/env/cocotb/cocotb/__init__.py", line 146, in _initialise_testbench
    regression.initialise()
  File "/userselec/chevillot/Projects/ATLAS/git/cocotb/atlas-lar-ldpb-firmware/env/cocotb/cocotb/regression.py", line 112, in initialise
    module = _my_import(module_name)
  File "/userselec/chevillot/Projects/ATLAS/git/cocotb/atlas-lar-ldpb-firmware/env/cocotb/cocotb/regression.py", line 63, in _my_import
    mod = __import__(name)
ImportError: No module named fpga_tb1
Cocotb initialisation failed - exiting
#      0.00ns INFO     cocotb.gpi                                  gpi_embed.c:237  in embed_sim_init                  Running on ModelSim for Questa-64 version 10.2c_5 2013.11
#      0.00ns INFO     cocotb.gpi                                  gpi_embed.c:238  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
#      0.00ns INFO     cocotb                                      __init__.py:109  in _initialise_testbench           Unable to determine Cocotb version from Unknown
#      0.00ns INFO     cocotb                                      __init__.py:128  in _initialise_testbench           Seeding Python random module with 1441619547
#      0.00ns ERROR    cocotb.scheduler                            __init__.py:169  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely

chiggs, 2015-09-11:
@chevillotn I suspect this might be to do with stdout.  On other simulators I get the following output:

```
run -all
# COUT:      0.00ns INFO     cocotb.gpi                                  gpi_embed.c:237  in embed_sim_init                  Running on Riviera-PRO version 2015.06.92.5791
# COUT:      0.00ns INFO     cocotb.gpi                                  gpi_embed.c:238  in embed_sim_init                  Python interpreter initialised and cocotb loaded!
# COUT:      0.00ns INFO     cocotb                                      __init__.py:112  in _initialise_testbench           Running tests with Cocotb v1.0 from /home/chiggs/code/github/cocotb
# COUT:      0.00ns INFO     cocotb                                      __init__.py:128  in _initialise_testbench           Seeding Python random module with 1441973746
# COUT: Traceback (most recent call last):
# COUT:   File "/home/chiggs/code/github/cocotb/cocotb/__init__.py", line 146, in _initialise_testbench
# COUT:     regression.initialise()
# COUT:   File "/home/chiggs/code/github/cocotb/cocotb/regression.py", line 114, in initialise
# COUT:     module = _my_import(module_name)
# COUT:   File "/home/chiggs/code/github/cocotb/cocotb/regression.py", line 63, in _my_import
# COUT:     mod = __import__(name)
# COUT: ImportError: No module named test_fail
# COUT: Cocotb initialisation failed - exiting
# KERNEL: Simulation has finished. There are no more test vectors to simulate.
endsim
# COUT:      0.00ns ERROR    cocotb.scheduler                            __init__.py:169  in _sim_event                      Failing test at simulator request before test run completion: Simulator shutdown prematurely
```

I suspect just logging an error via Python logging mechanism before raising the exception would improve visibility?

chevillotn, 2015-09-11:
I guess this was only my misunderstanding of where to look for in case of a problem. Thanks

chiggs, 2015-09-11:
@chevillotn anything that is unclear suggests it can be improved!  We work with Cocotb every day so it's easy to miss things that could improve the user experience.

I've made a fix that should provide a more useful log message which I'll merge now.  Thanks for pointing this out.

closed 2015-09-11 by chiggs