cocotb/cocotb #320: Cocotb with Chisel C++ model

Martoni, 2015-09-22:
Chisel is an HDL based on Scala language (Constructing Hardware In Scala Embedded Language).
Chisel can generate C++/SystemC model of the hardware description for simulation. This great to do fast simulation.
Chisel can also produce a verilog-syntetizable version of the hardware description for synthesis with all current constructor tools (ISE, Quartus, ...).
I'm using cocotb to simulate the verilog version of my Chisel design (with icarus), but it can be great to simulate the C++/SystemC with cocotb to speed the simulation.
Do you think it's a good idea to develop a "backend" for Chisel ?

stuarthodgson, 2015-09-22:
@Martoni I think this sounds like an interesting project. It all depends on what the interface to the C++ output looks like though, i'm curious so will have a look though the Chisel documentation. We would really want a way to plug into GPI to keep everything working.

Martoni, 2015-09-28:
Is there a documentation or an example code to know how to begin to do it ?

chiggs, 2015-09-29:
@Martoni the place to start would be to look at the `GpiImplInterface` class, which defines the behaviour that must be provided by an implementation.

An example for the VPI subclass that provides and implementation for VPI is here: https://github.com/potentialventures/cocotb/blob/master/lib/vpi/VpiImpl.h

I suppose the key question to answer would be the extent to which introspection is possible in SystemC (i.e. discovering signals by their name) and how that might be tackled.  Handles to hierarchy objects could just be pointers to System C class instances? I assume getting and setting values could be done simply by calling SystemC methods directly.

jahagirdar, 2019-01-07:
@chiggs  I have a client interested in this feature, and I can devote a week to working on this... Is there some guideline on what is required to interface cocotb with a new simulator?

Martoni, 2019-01-09:
It's an old thread, since this message chisel version was changed to Chisel 3.x.
And with chisel3, there is no more «C++ model», it convert hardware design to FIRRTL language then to Verilog language.
With chisel3, simulation is done with the excellent Verilator simulator.
Then to do fast simulation with chisel, the backend to add is Verilator !
It possible to do «slow» simulation with icarus, and it already works.

cmarqu, 2019-01-09:
@Nic30 has some Verilator binding with Python in https://github.com/Nic30/pycocotb FWIW

tsengr0916, 2019-01-10:
> I would like to make it 100% cocotb compatible (There are some special events which simplifies the programming of UVM like interface agents). It is scheduled after 20.1. I expect it would take cca 5 days. Also I will use it as main simulator for some libraries so there will be plenty of tests.

In our development, I have integrated the COCOTB and UVM TB under one sim environment, which is called PYCSV (Python/C/SV) hybrid framework. There will be two test benches running concurrently during a simulation, and we can reuse all the verification IPs, test sequences, and libraries from both COCOTB and UVM.

I have developed a PYCSV library suite, which is a reusable library consists of Python, C/C++, and SV code. It hides detail implementations of the runtime data exchange across language domains. It also handles the test flow  synchronization and start/stop processes.

The main test flow is driven from COCOTB, in which it can launch UVM sequences. Then an UVM sequence then can either be executed either on UVM side of VIP, or passes the randomized sequence item back to COCOTB, and to let COCOTB VIP to drive the stimulus.

It’s still in the review phase to our company internally. If someone is interested in this, we can talk off line.

Thanks,
Richard

Nic30, 2019-01-10:
@tsengr0916 Can I see at least some example of usage? It seems to me like you have sort of System Verilog compiler and Python wrapper for it. This kind of software tends to be quite complex. Is there any public doc for your project? (And there are some other project like that [slang](https://github.com/MikePopoloski/slang) ) Also which company do you represent? What do you expect from the result?

tsengr0916, 2019-01-10:
We use QuestaSim from Mentor 10.7b release to compile SV code and UVM TB. Here is the high level block diagram:
[cid:image001.png@01D4A8C4.B2485C10]

One example is we can parse and collect system model’s parameters in a Python/COCOTB test sequence, group those parameters in a string, and pass to UVM TB. So the flow path is like this: COCOTB sequence -> PYCSV in COCOTB -> pycsv_util.c -> pycsv_if.sv -> UVM coverage collector -> decompose the string and trig the coverage sampling. At the end of sim, the coverage database is built, and QuestasSim can read the database and provide GUI environment for coverage analysis and coverage hole debugging. This is the traditional SVTB coverage collection and analysis flow.  But the flow is more efficient and productive, and can probably be migrated to lab testing coverage collection without COCOTB.  The green boxes above are the PYCSV library I was talking about.

I work for Qualcomm. We use this framework/methodology to verify our 5G modem designs. There is no public web site to talk about this yet. It’s still in our review phase internally.

Richard

tsengr0916, 2019-01-10:
BTW, to answer Nic30's question, yes I use DPI to interface between C and SV code.

ktbarrett, 2020-04-23:
@tsengr0916 That seems quite interesting. Given your posts are now a year old perhaps you would like to post a final reflection in a new issue or on the gitter?

ktbarrett, 2020-04-23:
> It's an old thread, since this message chisel version was changed to Chisel 3.x.
> And with chisel3, there is no more «C++ model»

This seems like this issue is not valid anymore. If anyone is still interested in making a GP C++ model interface for cocotb, please make a new issue.

closed 2020-04-23 by ktbarrett

cmarqu, 2020-04-23:
@tsengr0916 Also, please take our 2020 cocotb survey (https://docs.google.com/forms/d/e/1FAIpQLSfD36PldzszbuZjysss3AMvxkf6XCtSbDTVh9qVNNYDaHTZ_w/viewform) unless you have already done so!