cocotb/cocotb #321: Install process on a windows 7(64 bits)

fdesla, 2015-09-22:
Hi everybody,

I am new to all of this and I'm facing problems installing everything. First of all, I am working on Windows 7 (64 bits) and I already have Pyhton 3.4.
To work with cocotb, I downloaded the minGWinstaller and checked the 3 packages as indicated on your datasheet, therefore I have now a console "MINGW32".
Then I downloaded "cocotb-0.3.zip", I unzipped it in the following paths ".\MinGW\msys\1.0" but when I'm doing the command "make" in the folder "cocotb-0.3" I have a lot of mistakes, starting by what is pasted below.
To be sure that what I understood is right: I do need to compile cocotb in order to use it therefore I have to use the "make" command.

Could somebody please explain to me what is wrong and what I have to do.

Thanks for advance
François

$ make
  File "<string>", line 1
    import os.path,distutils.sysconfig;print os.path.sep+os.path.join(*list(dist
utils.sysconfig.get_python_lib(standard_lib=True).split(os.path.sep)[:-1]))
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_version()
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
make -k -C examples
make[1]: Entering directory `/cocotb-0.3/examples'
  File "<string>", line 1
    import os.path,distutils.sysconfig;print os.path.sep+os.path.join(*list(dist
utils.sysconfig.get_python_lib(standard_lib=True).split(os.path.sep)[:-1]))
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_version()
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import os.path,distutils.sysconfig;print os.path.sep+os.path.join(*list(dist
utils.sysconfig.get_python_lib(standard_lib=True).split(os.path.sep)[:-1]))
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_version()
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
make[2]: Entering directory `/cocotb-0.3/examples/plusargs'
make -C /cocotb-0.3/lib/gpi
  File "<string>", line 1
    import os.path,distutils.sysconfig;print os.path.sep+os.path.join(*list(dist
utils.sysconfig.get_python_lib(standard_lib=True).split(os.path.sep)[:-1]))
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_version()
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
make[3]: python-config: Command not found
make[3]: Entering directory `/cocotb-0.3/lib/gpi'
gcc -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Waggregate-return -Wstrict
-prototypes -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -m32 -DFILTER -c
-fpic -I/cocotb-0.3/include -I  -o /cocotb-0.3/build/obj/i686/gpi_logging.o gpi_
logging.c
make[3]: gcc: Command not found
make[3]: *** [/cocotb-0.3/build/obj/i686/gpi_logging.o] Error 127
make[3]: Target `all' not remade because of errors.
make[3]: Leaving directory `/cocotb-0.3/lib/gpi'
make[2]: *** [/cocotb-0.3/build/libs/i686/libgpilog.so] Error 2
make -C /cocotb-0.3/lib/embed
  File "<string>", line 1
    import os.path,distutils.sysconfig;print os.path.sep+os.path.join(*list(dist
utils.sysconfig.get_python_lib(standard_lib=True).split(os.path.sep)[:-1]))
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_version()
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()

stuarthodgson, 2015-09-22:
@fdesla Did you really mean cocotb-0.3? If so then this does not work on windows, you need version 1.0 at least. You can get this either via [Cocotb-1.0](https://github.com/potentialventures/cocotb/archive/v1.0.zip).

You are correct that you need a working minGW environment, including gcc. Which from your later error does not seem to be there. The minGW tools page gives the steps needed to ensure that the environment is valid. [MinGW](http://www.mingw.org/wiki/howto_install_the_mingw_gcc_compiler_suite)

fdesla, 2015-09-22:
I downloaded through the website the 0.3 version... I tried with the 1.0 and I still have errors and when I'm looking to the MinGW installer I have :
[image]

So i do have the gcc compiler...

fdesla, 2015-09-22:
And thanks for your quick answer !!

stuarthodgson, 2015-09-22:
@fdesla It's not quite clear but from the screen shot above you have the C++ (g++) compiler but not C (gcc) - installed as part of 'mingw-gcc'. Both are required. My installer shows many more packages than your output above too.

Inside a mingw shell you should be able to type

$> which g++

and

$> which gcc

There should produce the same output showing where the compiler is located in the mingw installation.
If this fails (which I think that it will) then you may need to modify the PATH environment variable to include the location where the binary is located.

fdesla, 2015-09-22:
I don't have the same path but I have both see below :
[image]

But there is something I don't understand. I should be looking for a gnu python compiler, shouldn't I?
Thanks again for your help!

stuarthodgson, 2015-09-22:
@fdesla  The python compiler is simply "python" due to the language being interpreted. The build process with first build the libraries and then use python via the simulator to run a test. Since you have moved to the new version of cocotb (1.0) and changed your PATH can you post the output of the "make" command if you still have a problem.

fdesla, 2015-09-22:
Here is what I got after the "make" command on the MINGW32 console:

$ make
  File "<string>", line 1

And here is after the "python" command:

$ python
Python 3.4.3 (v3.4.3:9b73f1c3e601, Feb 24 2015, 22:43:06) [MSC v.1600 32 bit (In
tel)] on win32
Type "help", "copyright", "credits" or "license" for more information.

Thanks for your attention!
    from distutils import sysconfig; print sysconfig.get_config_var("DESTSHARED"
)
                                                   ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
make -k -C examples
make[1]: Entering directory `/cocotb-1.0/examples'
  File "<string>", line 1
    from distutils import sysconfig; print sysconfig.get_config_var("DESTSHARED"
)
                                                   ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    from distutils import sysconfig; print sysconfig.get_config_var("DESTSHARED"
)
                                                   ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
which: iverilog: unknown command
dirname: missing operand
Try `dirname --help' for more information.
make[2]: Entering directory `/cocotb-1.0/examples/plusargs'
/cocotb-1.0/makefiles/simulators/Makefile.icarus:43: *** "Directory containing I
carus binaries must be included in system path".  Stop.
make[2]: Leaving directory `/cocotb-1.0/examples/plusargs'
make[1]: *** [plusargs] Error 2
  File "<string>", line 1
    from distutils import sysconfig; print sysconfig.get_config_var("DESTSHARED"
)
                                                   ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    from distutils import sysconfig; print sysconfig.get_config_var("DESTSHARED"
)
                                                   ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
which: iverilog: unknown command
dirname: missing operand
Try `dirname --help' for more information.
[... 61 lines cut ...]
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
  File "<string>", line 1
    from distutils import sysconfig; print sysconfig.get_config_var("DESTSHARED"
)
                                                   ^
SyntaxError: invalid syntax
  File "<string>", line 1
    import distutils.sysconfig; print distutils.sysconfig.get_python_inc()
                                              ^
SyntaxError: invalid syntax
which: iverilog: unknown command
dirname: missing operand
Try `dirname --help' for more information.
make[2]: Entering directory `/cocotb-1.0/examples/endian_swapper/tests'
/cocotb-1.0/makefiles/simulators/Makefile.icarus:43: *** "Directory containing I
carus binaries must be included in system path".  Stop.
make[2]: Leaving directory `/cocotb-1.0/examples/endian_swapper/tests'
make[1]: *** [endian_swapper/tests] Error 2
make[1]: Target `all' not remade because of errors.
make[1]: Leaving directory `/cocotb-1.0/examples'
make: *** [test] Error 2

stuarthodgson, 2015-09-22:
@fdesla There is something a miss with your python installation. We have not tried 3.4 on windows but I have used 2.7. You are also missing the [icarus](http://iverilog.wikia.com/wiki/Installation_Guide) simulator that is the default choice if the SIM environment variable is not set. Once installed this also needs to be on your PATH.

fdesla, 2015-09-22:
Okay thanks again ;)

I'll go for the 2.7 version of python but I don't want to use icarus, I'm working on vivado (I was hoping it will be detected...)
And how do I know the name of "my" variable SIM?

chiggs, 2015-09-22:
@fdesla I'm not sure you'll have much success with Vivado.  When I last checked they hadn't implemented enough of a foreign language interface to support Cocotb.

imphil, 2019-01-24:
The original problem in this issue is resolved, closing.

closed 2019-01-24 by imphil