cocotb/cocotb #329: Building libfli fails

mkreider, 2015-09-30:
Hi folks,

I'm trying for a while now to get cocotb to use VHDL top files under Questasim on Ubuntu.
To my understanding, this requires the foreign language interface library, but I simply can't get cocotb's libfli.so to compile.

I changed the hardcoded include path in the FLI makefile to my installation of QS/include and forced building by setting BUILD_FLI.

Compile fails with the following output, any help is much appreciated

Best - Matt

```
...

make[4]: Entering directory `/home/mkreider/hdlprojects/cocotb/lib/fli'
g++ -Werror -Wcast-qual -Wcast-align -Wwrite-strings -Wall -Wno-unused-parameter -fno-common -g -DDEBUG -fpic -DFLI_CHECKING -c -fpic -I/home/mkreider/hdlprojects/cocotb/include -I/usr/include/python2.7 -I/opt/questa/questa_sv_afv_10.2c_5/questasim/include -o /home/mkreider/hdlprojects/cocotb/build/obj/x86_64/FliImpl.o FliImpl.cpp
In file included from FliImpl.cpp:30:0:
FliImpl.h: In constructor ‘FliSignalObjHdl::FliSignalObjHdl(GpiImplInterface*, mtiSignalIdT)’:
FliImpl.h:178:78: error: no matching function for call to ‘GpiSignalObjHdl::GpiSignalObjHdl(GpiImplInterface*&)’
FliImpl.h:178:78: note: candidates are:
In file included from FliImpl.h:31:0,
                 from FliImpl.cpp:30:
../gpi/gpi_priv.h:126:5: note: GpiSignalObjHdl::GpiSignalObjHdl(GpiImplInterface*, void*)
../gpi/gpi_priv.h:126:5: note:   candidate expects 2 arguments, 1 provided
../gpi/gpi_priv.h:124:7: note: GpiSignalObjHdl::GpiSignalObjHdl(const GpiSignalObjHdl&)
../gpi/gpi_priv.h:124:7: note:   no known conversion for argument 1 from ‘GpiImplInterface*’ to ‘const GpiSignalObjHdl&’
FliImpl.cpp: In member function ‘virtual GpiObjHdl* FliImpl::native_check_create(std::string&, GpiObjHdl*)’:
FliImpl.cpp:117:52: error: cannot allocate an object of abstract type ‘FliSignalObjHdl’
In file included from FliImpl.cpp:30:0:
FliImpl.h:173:7: note:   because the following virtual functions are pure within ‘FliSignalObjHdl’:
In file included from FliImpl.h:31:0,
                 from FliImpl.cpp:30:
../gpi/gpi_priv.h:102:24: note: 	virtual GpiObjHdl* GpiObjHdl::get_handle_by_name(std::string&)
../gpi/gpi_priv.h:103:24: note: 	virtual GpiObjHdl* GpiObjHdl::get_handle_by_index(uint32_t)
../gpi/gpi_priv.h:104:26: note: 	virtual GpiIterator* GpiObjHdl::iterate_handle(uint32_t)
../gpi/gpi_priv.h:105:24: note: 	virtual GpiObjHdl* GpiObjHdl::next_handle(GpiIterator*)
../gpi/gpi_priv.h:138:23: note: 	virtual GpiCbHdl* GpiSignalObjHdl::value_change_cb(unsigned int)
FliImpl.cpp: In member function ‘FliSignalCbHdl* FliSignalObjHdl::value_change_cb()’:
FliImpl.cpp:331:5: error: reference to ‘m_name’ is ambiguous
In file included from FliImpl.h:31:0,
                 from FliImpl.cpp:30:
../gpi/gpi_priv.h:115:17: error: candidates are: std::string GpiObjHdl::m_name
../gpi/gpi_priv.h:115:17: error:                 std::string GpiObjHdl::m_name
FliImpl.cpp:334:39: error: reference to ‘m_impl’ is ambiguous
In file included from FliImpl.h:31:0,
                 from FliImpl.cpp:30:
../gpi/gpi_priv.h:78:23: error: candidates are: GpiImplInterface* GpiHdl::m_impl
../gpi/gpi_priv.h:78:23: error:                 GpiImplInterface* GpiHdl::m_impl
FliImpl.cpp: In member function ‘GpiCbHdl* FliSignalObjHdl::rising_edge_cb()’:
FliImpl.cpp:342:5: error: reference to ‘m_name’ is ambiguous
In file included from FliImpl.h:31:0,
                 from FliImpl.cpp:30:
../gpi/gpi_priv.h:115:17: error: candidates are: std::string GpiObjHdl::m_name
../gpi/gpi_priv.h:115:17: error:                 std::string GpiObjHdl::m_name
FliImpl.cpp: In member function ‘GpiCbHdl* FliSignalObjHdl::falling_edge_cb()’:
FliImpl.cpp:348:5: error: reference to ‘m_name’ is ambiguous
In file included from FliImpl.h:31:0,
                 from FliImpl.cpp:30:
../gpi/gpi_priv.h:115:17: error: candidates are: std::string GpiObjHdl::m_name
../gpi/gpi_priv.h:115:17: error:                 std::string GpiObjHdl::m_name
make[4]: *** [/home/mkreider/hdlprojects/cocotb/build/obj/x86_64/FliImpl.o] Error 1
make[4]: Target `all' not remade because of errors.
```

stuarthodgson, 2015-09-30:
@mkreider 

Sorry you are having problems with the fli library. Which commit are you running?

Local tests on Linux Mint (ubuntu based) allow fli to compile against Modelsim, which uses the same Makefiles as questa on Master and v1.0. The fact that you have to set BUILD_FLI suggests you are on v1.0 though. This has since been fixed and FLI build is enabled on the master branch.

You should also not need to change the path in the FLI Makefile since this is now dynamic based on the location of 'vsim' on your PATH , this change went in between v0.4 and v1.0.

In summary can you confirm you are running on Master?

mkreider, 2015-09-30:
@stuarthodgson

Thank you, that did the trick! I feel a little stupid now for not trying to pull earlier, but since the option for FLI already existed I was convinced that the problem stemmed from my build environment, not your sources.

After pulling 836882, libfli builds without problems and after switching to 64 bit questa I can use vhdl topfiles. Thx again

stuarthodgson, 2015-09-30:
@mkreider 

Great, we are not great at doing releases I'm afriad but master is the best bet since that will get fixes quickly.

closed 2015-09-30 by stuarthodgson