cocotb/cocotb #330: Add support for records at top_level: vpiStructNet 683 object type

chevillotn, 2015-10-05:
I am using a signal of type record as top_level interface.
This type is not supported for the moment.

chevillotn, 2015-10-07:
The code is the following:

typedef struct
{
    logic a_in;
    logic b_out;
} test_if;

module module_tb_v(
    inout test_if inout_if
);
...
endmodule

In the python i use:
dut.inout_if.a_in = 0

This shouts with:
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:198  in gpi_get_handle_by_name          Checking if inout_if native though impl VPI 
#      0.00ns WARNING  cocotb.gpi                                  VpiImpl.cpp:106  in create_gpi_obj_from_handle      Not able to map type 683 to object.
#      0.00ns ERROR    cocotb.gpi                                  VpiImpl.cpp:130  in native_check_create             Unable to query fetch object module_tb_v.inout_if
#      0.00ns ERROR    cocotb.module_tb_v                            handle.py:107  in _raise_testerror                module_tb_v contains no object named inout_if

chiggs, 2015-10-07:
@chevillotn could you try off Issue#17 branch?  There's been a huge amount of work improving GPI recently so `master` is somewhat behind the times.

stuarthodgson, 2015-10-07:
@chevillotn 
Issue#17 branch does not handle vpiStructNet at the moment.

I'll try and get something done soon to support this though as should be pretty simple.

chevillotn, 2015-10-07:
sounds good! I hqve tried on my side for 1h to get it working but failed as i have no knowledge of VPI API :)

chevillotn, 2015-10-08:
Hi Stuart, do you think you could have something end of this week? i'd like to try it out :)

Thanks!

stuarthodgson, 2015-10-08:
@chevillotn Does not seem like it. I have done what I thought we be required, I think you are using Modelsim?

I can handle

```
dut.inout_if <= 0
```

But am not able to access any members of the structure. On one of my other simulators the structure does not show up as vpiStructNet but I can access the internals.

I have created a branch with the testcase on it but will not be able to continue until I finish some other work I am afraid.

chevillotn, 2015-10-08:
yes i am using modelsim (questa) v10.2c
any idea why you can't access the members of the structure?

I have also tried to use an interface to the testbench but modelsim does not allow me to simulate this module...

chevillotn, 2015-10-13:
Seems that i have to use something else than an inout structure. Any idea what i could use to accomplish the same? (split my structure into 2? one for in one for out?) or is there a chance i could get this working?

stuarthodgson, 2015-10-14:
@chevillotn this seems to be an issues with Modelsim as other simulators work as expected. The same code path is used regardless of the simulator. This is somewhat out of our control. I have tried using the newer Issue#17 Branch and still not not find any objects under a vpiStructNet. 

I would suggest either asking mentor for advice or splitting into two signals that can be accessed individually.

chevillotn, 2015-10-14:
I see... that's what i feared would be the case... I will try to contact Mentor with that. Maybe they'll tell me immediately that it is not supported :)
Thanks!

jahagirdar, 2018-06-13:
I find a similar issue with VHDL(ghdl),
I have a record at the toplevel IO and when I try to printout the interface 

 print "--------------"
    for thing in dut:
        print thing._fullname
    print "--------------"
I do not see the record.

csantosb, 2020-03-31:
Same here, using latest ghdl / cocotb

    make SIM=ghdl TOPLEVEL_LANG=vhdl

gives

    sample_module contains no object named inout_if

ktbarrett, 2020-05-05:
Support for `vpiStructNet` was added and top-level structures are accessible using teh VPI, so I'm closing this issue.

@jahagirdar @csantosb I'm continuing your issue for VHPI/GHDL support of top-level structures in #1770.

closed 2020-05-05 by ktbarrett