cocotb/cocotb #338: Wishbone BFMs

chiggs, 2015-11-08:
Issue to capture commits related to wishbone - already in progress on the `wishbone` branch by @mkreider.

mkreider, 2015-12-01:
Found some minor problems with the WB GPIO test design where it does not conform to WB B4 standard.

There is a problem with ack signal assignment

    //
    // WB Acknowledge
    //
    assign wb_ack = wb_cyc_i & wb_stb_i & !wb_err_o;

The internal signal wb_ack should not be depending on an output value. It should depend on the inputs 'wb_cyc_i', 'wb_stb_i', the internal signal 'wb_err', and, should you support it, the deasserted flow control signal '!wb_stall'.

And there is a problem with the error cases.
Part a.) is correct, non mapped addresses may result in an error.
Part b.), however, does not comply to the current WB standard.

    // All WISHBONE transfer terminations are successful except when:
    // a) full address decoding is enabled and address doesn't match
    //    any of the GPIO registers
    // b) wb_sel_i evaluation is enabled and one of the wb_sel_i inputs is zero

    //
    // WB Error
    //
    `ifdef GPIO_FULL_DECODE
    `ifdef GPIO_STRICT_32BIT_ACCESS
    assign wb_err = wb_cyc_i & wb_stb_i & (!full_decoding | (wb_sel_i != 4'b1111));
    `else
    assign wb_err = wb_cyc_i & wb_stb_i & !full_decoding;
    `endif
    `else
    `ifdef GPIO_STRICT_32BIT_ACCESS
    assign wb_err = wb_cyc_i & wb_stb_i & (wb_sel_i != 4'b1111);
    `else
    assign wb_err = 1'b0;
    `endif
    `endif

Check out WB B4, 3.5.4, Data Organization for 32-bit Ports.
If you want strict 32b access, you can change the interface to have only 1 select line.
If you want 32 and 16b word access, you can change the interface to have 2 select lines.
If you want 32, 16 and 8, you have 4 byte select lines.

But you should not superimpose your own rules on the use of the existing select lines. It is perfectly okay if one or all select lines are '0'. Declaring it an error if they not all show '1' is not compliant to the standard. 

Cheers - Matt

ktbarrett, 2021-01-30:
Closing duplicate of #862.

closed 2021-01-30 by ktbarrett