cocotb/cocotb #341: Makefile with Riviera-Pro 2015.10

saturn77, 2015-11-19:
Hello All;   I have cocotb running fine with "make" for all the examples, but not when I run "make SIM=aldec" and the message that I get is : 

dirname: missing operand
Try 'dirname --help' for more information.
/home/james/public_sw/cocotb/makefiles/simulators/Makefile.aldec:96: *** "Directory containing Riviera-Pro binaries must be included in system path".  Stop.

However, in my .bashrc file I have the path including the Riviera-Pro bin directory. 

Any suggestion on how to get this to work is much appreciated. -James

chiggs, 2015-11-19:
@saturn77 the Makefile calls `which vsim` to check where the Aldec simulator is installed.  What happens if you run the command, is it found?

Can you list the contents of the `$riviera/bin` directory.  I haven't installed 2015.10 yet so perhaps they've changed the installation layout in the new release.

Technically we should use `which riviera` anyway since vsim is just an alias to make the Aldec tool appear more like Modelsim.

saturn77, 2015-11-19:
So when I call "which vsim" and "which riviera" I get the following results.....

[image]

The contents of the /bin directory are shown here: 

[image]

saturn77, 2015-11-20:
Chris, any suggestions to make changes to  
/cocotb/makefiles/simulators/Makefile.aldec file ? 

Thanks, James

chiggs, 2015-11-20:
@saturn77 one thing you could do is comment out line 93, and set the environment variable `ALDEC_BIN_DIR=/home/james/riviera2015.10/bin/Linux64`, which should workaround this.

Also if you could post the shell you're using that would be helpful, in addition to the output of the following:

```
dirname `which riviera`
```

saturn77, 2015-11-20:
When running dirname which riviera or dirname 'which riviera' I get the following, and I'm running bash shell.

[image]

chiggs, 2015-11-20:
Note the backticks rather than single quotes, which instructs bash to run the command in ticks first and then use the output as the argument to dirname.

saturn77, 2015-11-20:
Sorry about that, yes, it seems to be pointing correctly. 

[image]

stuarthodgson, 2016-02-05:
closing as no update for 3 months

closed 2016-02-05 by stuarthodgson