cocotb/cocotb #343: Lack of simulation time unit information makes writing portable code hard.

luked80, 2015-11-25:
Code that controls the timing of a signal on a certain interface might look something like this:
```python
dut.VDD <= 0
yield Timer(1)
dut.VDD <= 1
yield Timer(100)
yield do_test(dut)
```

VDD controls an internal power up reset circuit that takes 100ns to come out of reset.
This code will work fine if the timescale of the simulation is 1ns/1ns.
If I introduce a piece of ip with a smaller timescale or modify the timescale for some other reason, I need to go go back and rewrite the code.

Giving cocotb some idea of time units would really help here.

A set of convenience functions that map timing info into simulation counts would be one way to go.
The above code would look something like:
```python
dut.VDD <= 0
yield Timer(ns(1))
dut.VDD <= 1
yield Timer(ns(100))
yield do_test(dut)
```

The existing cocotb logging timestamps also suffer from changing timescales.

It's possible to get the simulation timescale precision from the vpi with:
```c
precision = vpi_get(vpiTimePrecision, NULL);
```

chiggs, 2015-11-27:
Completely agree - see also #115 and #289.

luked80, 2016-03-07:
This is now being covered in #115

closed 2016-03-07 by luked80