cocotb/cocotb #344: Call to "hasattr" generate log warning if attribute doesn't exist

Martoni, 2015-12-01:
In avalon driver (cocotb/drivers/avalon.py), there is some call to "hasattr()"  to know if an optional signal exists.

But if doesn't exist, a warning message is generated :  
```
Failed to find a hdl named %s via any registered implementation
```
This warning is generated in C function gpi_get_handle_by_name() in lib/gpi/GpiCommon.cpp file. Can we change it to not generate warning when hasattr() is called ?

stuarthodgson, 2015-12-01:
@Martoni , There is no way for the C++ code to know that the call has come from hasattr. We could remove the the warning all together and leave to the Python to handle.

Martoni, 2015-12-01:
@stuarthodgson, that should be better to remove it no ? because it's not really a warning in that case.

Martoni, 2015-12-01:
I suggest to transform "LOG_WARN" in "LOG_DEBUG" : https://github.com/potentialventures/cocotb/pull/345

closed 2015-12-01 by stuarthodgson