cocotb/cocotb #348: More than one yield Edge on the same signal does not work

chevillotn, 2015-12-11:
I am driving a signal dut.reset_n to toggle '1' -> '0' -> '1'
I have created a class TBWire that will duplicate this signal, i.e.
waits on an edge from the input signal and drives the output signal.

I instantiate 2 times this class on the same input signal dut.reset_n to drive dut.mmc_reset_n and dut.lli_global_rst
Only the last registered output signal is driven dut.lli_global_rst.
It seems the first registered edge is ignored...
Can we have multiple Edge classes on the same signal?

[test_cocotb.py.txt](https://github.com/potentialventures/cocotb/files/59419/test_cocotb.py.txt)

chevillotn, 2015-12-11:
From this test, output log shows:
#      0.00ns INFO     cocotb.TBOneShot.reset_n                   testbench.py:31   in _start                          Starting one-shot "Global asynchronous reset"
#      0.00ns INFO     cocotb.TBWire.reset_n.mmc_reset_n          testbench.py:55   in __init__                        Starting TBWire: source: reset_n @0x3264810 destination: mmc_reset_n @0x39427b0
#      0.00ns INFO     cocotb.TBWire.reset_n.mmc_reset_n          testbench.py:60   in _start                          TBWire: waiting edge on source: reset_n @0x3264810=0 destination: mmc_reset_n @0x39427b0
#      0.00ns INFO     ..otb.TBWire.reset_n.lli_global_rst        testbench.py:55   in __init__                        Starting TBWire: source: reset_n @0x3264810 destination: lli_global_rst @0x396a220
#      0.00ns INFO     ..otb.TBWire.reset_n.lli_global_rst        testbench.py:60   in _start                          TBWire: waiting edge on source: reset_n @0x3264810=0 destination: lli_global_rst @0x396a220
#      0.00ns INFO     ..otb.TBWire.reset_n.lli_global_rst        testbench.py:63   in _start                          TBWire: edge on source: reset_n @0x3264810=1 destination: lli_global_rst @0x396a220=1
#      0.00ns INFO     ..otb.TBWire.reset_n.lli_global_rst        testbench.py:60   in _start                          TBWire: waiting edge on source: reset_n @0x3264810=1 destination: lli_global_rst @0x396a220
#     10.00ns INFO     ..otb.TBWire.reset_n.lli_global_rst        testbench.py:63   in _start                          TBWire: edge on source: reset_n @0x3264810=0 destination: lli_global_rst @0x396a220=0
#     10.00ns INFO     ..otb.TBWire.reset_n.lli_global_rst        testbench.py:60   in _start                          TBWire: waiting edge on source: reset_n @0x3264810=0 destination: lli_global_rst @0x396a220
#    510.00ns INFO     ..otb.TBWire.reset_n.lli_global_rst        testbench.py:63   in _start                          TBWire: edge on source: reset_n @0x3264810=1 destination: lli_global_rst @0x396a220=1
#    510.00ns INFO     ..otb.TBWire.reset_n.lli_global_rst        testbench.py:60   in _start                          TBWire: waiting edge on source: reset_n @0x3264810=1 destination: lli_global_rst @0x396a220

lli_global_rst is correctly driven upon reset_n edge.
However mmc_reset_n is not driven... and class is still waiting for an edge..

chevillotn, 2015-12-11:
Looking at the gpi_register_value_change_callback function, it is clear that it overwrites existing registered function if called twice.
There should be an array of registered function/edge for each signal.

stuarthodgson, 2015-12-14:
@chevillotn , this does not seem to be the error that you think it is. For instance using a RisingEdge/FallingEdge works fine and these call the same underlying gpi function as Edge.  Edge though is not a singleton in triggers.py.

chevillotn, 2015-12-14:
Have you tried using 2 falling edges? or 2 rising edges? I think it will do the same, i.e. only one of them will be notified.

stuarthodgson, 2015-12-14:
@chevillotn , yes I have a testcase that does this and they work fine. I'm working on a fix around the day job so should have something this evening GMT.

chevillotn, 2015-12-14:
Ah sounds good, when you have something let me know which branch/testcase i can check on my side!
Thanks

stuarthodgson, 2015-12-15:
@chevillotn have a look at c8f583c testcase passed now.

chevillotn, 2015-12-15:
I can confirm that Edge testcase #348 fails if not integrating changes in the cocotb library.
The 2 testcases pass when cocotb changes are integrated.
Also my testnbench works as expected with integrated changes.

Thanks a lot!

closed 2015-12-15 by chevillotn