cocotb/cocotb #354: Support for 64-bit Python on Windows

jmorarity, 2015-12-15:
With Xilinx slowly deprecating x32 support and ModelSim DE finally providing x64 support (!), I started working at getting a build to run on Windows in 64-bit mode. I was ultimately successful using msys2 w/  mingw64 but had to make a few changes locally to the cocotb project. Unless this work is superseded by another effort I'd like to put together a pull request for it.

stuarthodgson, 2015-12-15:
@jmorarity . We are not working on this ourselves and no other contributors have mentioned it either. I've only worked on 32 bit windows so far as well but will try to help where I can, look forward to your pull requests.

jmorarity, 2016-01-06:
The "Windows 7 installation" section of the quickstart guide should be amended to add a 64bit section.  The existing 32-bit MinGW/msys procedure is still valid, but it should be noted that w/ MinGW-w64/msys2 you can run either 32 or 64 bit simulations (so it supersedes in a way).

The most basic documentation to add would be to install msys2-x86_64 (comes w/ mingw32 and mingw64) via the instructions at https://msys2.github.io/ and then follow the "building packages" guide at http://sourceforge.net/p/msys2/wiki/Contributing%20to%20MSYS2/ to acquire the requisite toolchains.
Finally, you must install the gcc toolchains

closed 2020-04-09 by eric-wieser