cocotb/cocotb #359: Array of Array of std_logic incorrectly seen in python scripts

chevillotn, 2015-12-16:
The following signals are defined in the top level sv file:
    wire dbg_test_array_1_std_logic[0:0];
    wire dbg_test_array_4_std_logic[3:0];
    wire dbg_test_array_1_array_8_std_logic[0:0][7:0];
    wire dbg_test_array_8_array_1_std_logic[7:0][0:0];

They are seen as expected in modelsim, see attached picture.

The python cocotb test is trying to access those signals:
    print("dbg_test_array_1_std_logic: %s" % (dut.dbg_test_array_1_std_logic))
    print("dbg_test_array_4_std_logic: %s" % (dut.dbg_test_array_4_std_logic))
    print("dbg_test_array_1_array_8_std_logic: %s" % (dut.dbg_test_array_1_array_8_std_logic))
    print("dbg_test_array_8_array_1_std_logic: %s" % (dut.dbg_test_array_8_array_1_std_logic))

The following debug output is seen:
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:241  in __gpi_get_handle_by_name        Searching for dbg_test_array_1_std_logic
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:254  in __gpi_get_handle_by_name        Checking if dbg_test_array_1_std_logic native though impl VPI
#      0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:131  in initialise                      VPI: dbg_test_array_1_std_logic initialised with 1 elements
#      0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:196  in create_gpi_obj_from_handle      VPI: Created object with type was vpiNetArray(114)
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:264  in __gpi_get_handle_by_name        Found dbg_test_array_1_std_logic via VPI
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:50   in check_and_store                 Checking fpga_tb_v.dbg_test_array_1_std_logic exists
#      0.00ns DEBUG    cocotb.dbg_test_array_1_std_logic             handle.py:90   in __init__                        Created
# dbg_test_array_1_std_logic: dbg_test_array_1_std_logic @0x39cc350
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:241  in __gpi_get_handle_by_name        Searching for dbg_test_array_4_std_logic
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:254  in __gpi_get_handle_by_name        Checking if dbg_test_array_4_std_logic native though impl VPI
#      0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:131  in initialise                      VPI: dbg_test_array_4_std_logic initialised with 4 elements
#      0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:196  in create_gpi_obj_from_handle      VPI: Created object with type was vpiNetArray(114)
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:264  in __gpi_get_handle_by_name        Found dbg_test_array_4_std_logic via VPI
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:50   in check_and_store                 Checking fpga_tb_v.dbg_test_array_4_std_logic exists
#      0.00ns DEBUG    cocotb.dbg_test_array_4_std_logic             handle.py:90   in __init__                        Created
# dbg_test_array_4_std_logic: dbg_test_array_4_std_logic @0x39cd2d0
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:241  in __gpi_get_handle_by_name        Searching for dbg_test_array_1_array_8_std_logic
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:254  in __gpi_get_handle_by_name        Checking if dbg_test_array_1_array_8_std_logic native though impl VPI
#      0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:131  in initialise                      VPI: dbg_test_array_1_array_8_std_logic initialised with 8 elements
#      0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:196  in create_gpi_obj_from_handle      VPI: Created object with type was vpiNetArray(114)
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:264  in __gpi_get_handle_by_name        Found dbg_test_array_1_array_8_std_logic via VPI
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:50   in check_and_store                 Checking fpga_tb_v.dbg_test_array_1_array_8_std_logic exists
#      0.00ns DEBUG    ..bg_test_array_1_array_8_std_logic           handle.py:90   in __init__                        Created
# dbg_test_array_1_array_8_std_logic: dbg_test_array_1_array_8_std_logic @0x39cde60
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:241  in __gpi_get_handle_by_name        Searching for dbg_test_array_8_array_1_std_logic
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:254  in __gpi_get_handle_by_name        Checking if dbg_test_array_8_array_1_std_logic native though impl VPI
#      0.00ns DEBUG    cocotb.gpi                                 VpiCbHdl.cpp:131  in initialise                      VPI: dbg_test_array_8_array_1_std_logic initialised with 8 elements
#      0.00ns DEBUG    cocotb.gpi                                  VpiImpl.cpp:196  in create_gpi_obj_from_handle      VPI: Created object with type was vpiNetArray(114)
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:264  in __gpi_get_handle_by_name        Found dbg_test_array_8_array_1_std_logic via VPI
#      0.00ns DEBUG    cocotb.gpi                                GpiCommon.cpp:50   in check_and_store                 Checking fpga_tb_v.dbg_test_array_8_array_1_std_logic exists
#      0.00ns DEBUG    ..bg_test_array_8_array_1_std_logic           handle.py:90   in __init__                        Created

It can be seen that both dbg_test_array_1_array_8_std_logic and dbg_test_array_8_array_1_std_logic are seen as arrays of size 8... however dbg_test_array_1_array_8_std_logic is an array of size 1 of an array of size 8. So it should be seen as 1 element and not 8.

[image]

chiggs, 2016-03-01:
See also #393

leftink, 2016-04-07:
@chevillotn Does this work correctly now that #393 has been merged in?

chevillotn, 2016-05-03:
I have integrated latest changes as mentioned.
I can see now the behavior is correct:
# DBG10: len(dbg_test_array_1_std_logic)=1
#     DBG11: len(dbg_test_array_1_std_logic[0])=1
# DBG10: len(dbg_test_array_4_std_logic)=4
#     DBG11: len(dbg_test_array_4_std_logic[0])=1
#     DBG11: len(dbg_test_array_4_std_logic[1])=1
#     DBG11: len(dbg_test_array_4_std_logic[2])=1
#     DBG11: len(dbg_test_array_4_std_logic[3])=1
# DBG10: len(dbg_test_array_1_array_8_std_logic)=1
#     DBG11: len(dbg_test_array_1_array_8_std_logic[0])=8
# DBG10: len(dbg_test_array_8_array_1_std_logic)=8
#     DBG11: len(dbg_test_array_8_array_1_std_logic[0])=1
#     DBG11: len(dbg_test_array_8_array_1_std_logic[1])=1
#     DBG11: len(dbg_test_array_8_array_1_std_logic[2])=1
#     DBG11: len(dbg_test_array_8_array_1_std_logic[3])=1
#     DBG11: len(dbg_test_array_8_array_1_std_logic[4])=1
#     DBG11: len(dbg_test_array_8_array_1_std_logic[5])=1
#     DBG11: len(dbg_test_array_8_array_1_std_logic[6])=1
#     DBG11: len(dbg_test_array_8_array_1_std_logic[7])=1

Also i have tried to set the signals one by one and it works well as shown on picture:
[image]

Thanks a lot! I close the ticket then :)

closed 2016-05-03 by chevillotn