cocotb/cocotb #368: LOG_ERROR() does not result in Test Error

leftink, 2015-12-20:
I was running a test that exercised some features that were not supported yet in the FLI interface.

For instance, the FLI interface for get_signal_value_long() had not been implemented and called LOG_ERROR() and then returns "-1".  Unfortunately, the test set the signal to -1 and then tried to read the signal which also returned "-1".  The test reported as a "PASS", but seems like it should have been a "FAIL".

@chiggs Is this desired behavior or should LOG_ERROR() also trigger a Test Error?

stuarthodgson, 2015-12-21:
@leftink, We do not return any error condition from any of the "get" functions in GPI I'm tempted to update this and pass the raising of an TestError up to python.

leftink, 2015-12-21:
@stuarthodgson, that might be a good idea.  The "get" functions was just an example.  It seems like any LOG_ERROR() that occurs should make sure the test is marked as a failure as well.

ktbarrett, 2020-12-18:
The best course of action is to bubble exceptions up to Python. This could end the test or be handled, depending on the type of error. Duplicated by #1844.

closed 2020-12-18 by ktbarrett