cocotb/cocotb #371: In Handle.py, recommend moving __len__() function to the SimBaseHandle class

leftink, 2016-01-04:
Currently, only "NonConstantObject" handles support the __len__() function.  This means in tests have to be careful of the handle type before invoking the len(handle) function.  This function will have the same implementation for all handle types so if we just move to the SimBaseHandle class, it will at least not throw an exception if someone calls len() on a handle.  The GPI interface already supports this for all handles and defaults returning 0, so it will not crash even if the capability for particular handle types is not supported by the different interfaces.

stuarthodgson, 2016-01-04:
@leftink makes sense

leftink, 2016-04-01:
Fixed in latest.  Closing.

closed 2016-04-01 by leftink