cocotb/cocotb #375: compiling FLI library with Questa 10.2c fails

chevillotn, 2016-01-07:
FliImpl.cpp: In function ‘void fli_mappings(GpiIteratorMapping<int, FliIterator::OneToMany>&)’:
FliImpl.cpp:517: error: ‘accElsifGenerate’ was not declared in this scope
FliImpl.cpp:518: error: ‘accElseGenerate’ was not declared in this scope
FliImpl.cpp:520: error: ‘accCaseGenerate’ was not declared in this scope
FliImpl.cpp:521: error: ‘accCaseOTHERSGenerate’ was not declared in this scope

Those IDs are not defined in the Questa v10.2c

The VHDL_LAST_ACC_ID defined in acc_vhdl.h should be checked.
For Questa v10.2c it is defined to accForeignObject

I guess accElsifGenerate is above this ID not defined in 10.2c.

stuarthodgson, 2016-01-09:
@chevillotn , alas this is not the case. The defines between versions of acc_vhdl.h arbitrarily change. for instance 

10.1c
```
#define    accForGenerate             1032
#define    accPackage                 1040
#define    accPackageUninstantiated   1041
```

10.3c
```
#define    accElsifGenerate           1032
#define    accElseGenerate            1033
#define    accForGenerate             1034
#define    accCaseGenerate            1035
#define    accCaseOTHERSGenerate      1036
#define    accPackage                 1040
#define    accPackageUninstantiated   1041
```

just checking VHDL_LAST_ACC_ID is not enough.

stuarthodgson, 2016-01-09:
@chevillotn does 2409ffb.. resolve this for you?

leftink, 2016-01-11:
@stuarthodgson I was able to roll back to Questa 10.2c and build and reproduce the failure above.  I then tried 2409ffb.. and it fixed the issue.

That was my bad, those failures come from VHDL-2008 support in Questa.  There are just too many versions of these tools.  I really wish we didn't have to use the acc_vhdl.h file, but the FLI interface just doesn't provide enough information.

stuarthodgson, 2016-01-11:
@leftink , the define space had room why they changed it is beyond me.

closed 2016-01-11 by stuarthodgson

chevillotn, 2016-01-12:
Ah yes indeed quite a lot of different versions! The fix solved the issue.
I see this FLI is new compared to what i integrated before. Does that mean we can use VHDL testbenches directly?